ArduPilot/MissionPlanner #1184: Create KML + gpx broken in new updates

SlipstreamFPV, 2016-03-11:
Since updating my mission planner software I can no longer export KML and gpx files from my logs. I tried new files and tried to recreate files I've processed previously and they also didn't work. I have also tried on multiple systems. Maybe something was accidentally broken in the newer updates.

meee1, 2016-03-11:
can you please post the tlog. is there anything odd about your setup?

SlipstreamFPV, 2016-03-11:
Thanks for your response. It's actually a .bin straight from the pixhawk memory card converted using mission planner to .log then attempted to do the gpx and kml conversion. I did test some old files again which are much smaller than this file and they actually did convert fine. Maybe it's just the size that's the issue. The .log file is just over 300mb. It crashes when it generates the .kml.tmp file on both of my computers. The .gpx file is finished properly before this stage just not the kml.

meee1, 2016-03-11:
300mb should be ok, I have tested with larger, for this exact reason.
I didn't know it was a .bin log though.

what firmware created the log?

SlipstreamFPV, 2016-03-11:
APM Plane 3.5. I did convert some other APM Plane 3.5 logs that are smaller successfully. Would it help if I upload the .bin for you to try?

meee1, 2016-03-11:
if you can upload the start of the log, that causes the issue that would be helpful.
NOTE it must be from the start of the file. the start contains fmt messages that are required to decode the .bin

SlipstreamFPV, 2016-03-11:
https://www.dropbox.com/s/jidzlr0b6d3i4tb/11%20CUT.log?dl=0

SlipstreamFPV, 2016-03-11:
That file is the first 1/4 of the .log file which mission planner made from the .bin

meee1, 2016-03-11:
I meant the .bin, ie I need the part that crashes.

SlipstreamFPV, 2016-03-12:
It crashes if it's the .bin or the .log that I load. The conversion from bin to log works just fine. Just when it tries to makes the actual gpx and kml files it crashes.

meee1, 2016-03-12:
ok, I downloaded the file you uploaded, and it process's fine for me.
can you try MP beta?

help > update beta

SlipstreamFPV, 2016-03-12:
Beta version processes it just fine! Thanks for the follow-ups.Looks like whatever the issue was it's fixed already :) Keep up the amazing work!

closed 2016-03-12 by meee1