ArduPilot/MissionPlanner #1185: Helicopter SITL doesn't take off

Fortesque0, 2016-03-11:
Hello,

Helicopter on sitl environment doesn't take off. You can see ciclyc servos increasing value (ch1,ch2 and ch3 output), but helicopter doesn't move. (I test it assign the throttle on both channels 3 and 8,without satisfactory results).
Maybe this issue is because the heli firmware throttle output is through ch8.
Thanks

rmackay9, 2016-03-14:
Don't think this is a mission planner issue.  It probably belongs in the ardupilot issues list.
But in any case, I've used the helicopter sitl in the last couple of weeks and it does work.  The sequence is:
rc 3 1000
rc 8 1000
arm throttle
rc 8 2000
rc 3 1800
and it should take off.

Fortesque0, 2016-03-14:
Hello Mackay9,
Yes, you can use it last weeks, but, seems that new release may contain bugs that did not exist in previous versions.
SITL on Helicopter and Airplane doen´t work fine on 1.35 and 1.36 MP versions. Airplane its moving always from start, without throttle and arming, and helicopter do the same, but when you lands and disarme. I can upload  logs if necesary.
Please, try to check it before to close the issue. 
Thanks

magicrub, 2016-03-14:
have you tried "sim_vehicle.sh -w" to wipe your eeprom? Maybe you have a
setting in there goofing things up. Works for me on v1.3.36

Fortesque0, 2016-03-14:
Hello Magicrub,
I have not used the command line. I do not know how it works. One of the reasons why I chose MP was for its graphical interface, avoiding having to make command lines. But if I'm not mistaken, the SITL is downloaded from the repository if it is not installed on your computer, and that's what I do every new version of MP, delete all files for the installation is as clean as possible. That means that on a new installation, MP generates SITL folder and downloads the firmware from repository. Doing that, SITL on Helicopter and Airplane fails (Copter its fine and Rover test pending:) ). I will attach log files as soon as possible.
Thanks

Fortesque0, 2016-03-14:
Hello,
The log files for both test, helicopter and plane. Heli with misión and plane just sitl loaded. 

[Airplane, Heli Logs  2016-03-14 19-50-26.zip](https://github.com/diydrones/MissionPlanner/files/172683/Airplane.Heli.Logs.2016-03-14.19-50-26.zip)

Fortesque0, 2016-03-19:
Hello,
Any idea of what happends?

Fortesque0, 2016-03-28:
Nobody have test it?

closed 2017-07-13 by Fortesque0

mohammad361, 2021-06-18:
same problem with heli sitl, arming, but cant take off

rmackay9, 2021-06-21:
I've tested in regular SITL (no MP SITL) and it is possible to get the helicopter flying.

I followed the same instructions as I listed above.
rc 3 1000
rc 8 1000
arm throttle
rc 8 2000
rc 3 1800

I had to wait a little while after the last "rc 3 1800" because it takes some time for the main rotor to get up to full speed.

The problem with trying to use TradHeli simulator in MP is that there is no way to move the interlock switch unless you have a joystick.

mohammad361, 2021-06-21:
hi, thanks a lot.

> I've tested in regular SITL (no MP SITL) and it is possible to get the helicopter flying.
>
> I followed the same instructions as I listed above.
> rc 3 1000
> rc 8 1000
> arm throttle
> rc 8 2000
> rc 3 1800
>
> I [...]

hi, thanks a lot.