ArduPilot/MissionPlanner #1188: EKF and Vibe in HUD can not be turned back on

Jman841, 2016-03-13:
The latest version of Mission Planner states it allows you to add and remove the EKF and Vibe information in the HUD, however, I can not find anywhere in mission planner to Turn this feature on/off.

meee1, 2016-03-13:
its an internal function. its not meant for general users

closed 2016-03-13 by meee1

Jman841, 2016-03-13:
I find these indicators very useful, especially with Quadcopters and helicopters as I can check to see if there is clipping issues with the pixhawk when building a new quadcopter and check vibrations easily during flight. And I can see if there is compass errors right away with the EKF indicator.

Is there any way we can make these optional to be used/visible in the HUD again?

meee1, 2016-03-13:
I'm fixing atm, they are meant to be on. you will get another update request very soon.

Jman841, 2016-03-13:
Ah ok, Thank you very much. Sorry for the misunderstanding.

meee1, 2016-03-13:
thanks for the report!!!