ArduPilot/MissionPlanner #1189: Survey enhancement request for Planes

jcooper21, 2016-03-14:
It would be nice if when planning a survey mission in the advanced tab there was an option for some defaults for a plane survey such as having another waypoint automatically dropped at the first waypoint location for the plane to circle in order to get proper wind speed reading prior to mission starting. 

Another nice feature would be for it to automatically drop the do start land command for planes. Since the location of due start land does not matter this could be dropped at the home point and would eliminate a step from mission planning.

magicrub, 2016-03-14:
You have a lot of requests, I hope they get done but the work load is
pulling up. Have you ever considered hiring a developer to get these done
for you in a timely manner?

jcooper21, 2016-03-14:
I was just posting the few things that I have noticed over the last while that can improve user experience or help eliminate bugs. Ultimately, they get done when and if they get done but hopefully it brings future attention to it.

jcooper21, 2016-03-14:
I should also note these posts are a result of the developer community requesting us to test and provide feedback on the software. If it's an issue I am more than happy to stop dedicating staff to go test software to provide feedback on ways to improve it for the end user.

magicrub, 2016-03-14:
We certainly appreciate your testing and the feedback is important to us, thank you.