ArduPilot/MissionPlanner #1193: Geo Ref Images error

shaggee, 2016-03-16:
Error while trying to geotag images:

`Using AMSL Altitude True
Reading log for GPS Messages in order to get AMSL Altitude
Log Read for GPS Messages
Reading log for CAM Messages
Error System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at MissionPlanner.GeoRef.Georefimage.readCAMMsgInLog(String fn) in c:\Users\michael\Source\Repos\MissionPlanner\GeoRef\georefimage.cs:line 355
   at MissionPlanner.GeoRef.Georefimage.doworkCAM(String logFile, String dirWithImages) in c:\Users\michael\Source\Repos\MissionPlanner\GeoRef\georefimage.cs:line 922
   at MissionPlanner.GeoRef.Georefimage.BUT_doit_Click(Object sender, EventArgs e) in c:\Users\michael\Source\Repos\MissionPlanner\GeoRef\georefimage.cs:line 1435`

Newest MP version
APM 2.6, Arducopter 3.1.5 simple mission with Do_Digicam_Control commands

[2016-03-16 14-44-41.zip](https://github.com/diydrones/MissionPlanner/files/176290/2016-03-16.14-44-41.zip)
[image]

meee1, 2016-03-16:
the log file and firmware you are using is to old.
the cam message is missing the expected relative alt value/field.
I would suggest you upgrade.

shaggee, 2016-03-16:
Thank you.
So I have to use arducopter 3.2 instead of 3.1.5 right?

meee1, 2016-03-25:
yes

closed 2016-03-25 by meee1