ArduPilot/MissionPlanner #1197: Match Frame Type in Motor Test Window

davidbitton, 2016-03-21:
It would make a lot more sense if the Motor Test buttons mimicked what is shown in the Frame Type window. For example, clockwise from front left (looking down at the craft), the motors are 1, 3, 2, and 4 but the motor test buttons are A, B, C and D. 

The image from the Frame Type window:

[image]

would be used for the Motor Test window:

[image]

With motor _numbers_ and not _letters_.

meee1, 2016-03-21:
the reason its done like this is because MP isn't able to tell exactly what frame type its connected to.
until changes are done in arducopter this wont change.

meee1, 2016-03-25:
its also why there is a link to the wiki page.