ArduPilot/MissionPlanner #1199: Storm 32 gimbal can't change modes

Jman841, 2016-03-23:
When using the Mavlink connection on a Storm32 gimbal. If you choose "Point Camera Here" in the map, You can't change back to RC Targeting when using the joystick.

meee1, 2016-03-25:
I'm not sure this is a MP issue.
are you changing the mount back to rc targeting?

Jman841, 2016-03-28:
That is the issue. Once it changes to the GPS pointing or I choose RC Targeting and select "Set Mount" but it does not do anything.

Also, with the latest version of Mission Planner, I can not even select modes and the mode shown in the GUI with Arducopter is "Unknown". I believe this is a separate issue though.

meee1, 2016-04-02:
are you selecting the gimbal before you do the set mount from the menu up the top right?

gediminasgu, 2016-09-21:
As I see from ardupilot code https://github.com/ArduPilot/ardupilot/blob/b8011fd46cc97b0e8161b35744328c1319beb0f0/APMrover2/commands_logic.cpp#L98
The mode will be reset to default (where default is ``restores the mode to it's default mode held in the MNT_DEFLT_MODE parameter``) if you will send lat/lng/alt as 0/0/0. And you can do that choosing from menu "Point camera to coords" and entering "0;0;0".

rmackay9, 2023-01-30:
this is a really old issue so closing.  I'm sure it is possible to set the Mount's mode with MP these days.

closed 2023-01-30 by rmackay9