ArduPilot/MissionPlanner #1210: Show indication that camera is triggering

iskess, 2016-03-31:
Use MAVLINK_MSG_ID_CAMERA_FEEDBACK to provide an indication that the mapping camera is triggering as expected. This could be an icon on the HUD, or just a HUD user_item. 

To take this a step further, show the time interval between triggers. Using distance based triggering, it is useful to know how the wind is affecting your triggering rate. If the rate gets too high due to a tailwind, the operator may consider slowing down the aircraft, increasing the altitude, or aborting the mission. In any case it's very helpful to be made aware that there may be a problem with your overlap.

meee1, 2016-04-02:
do you have a log you could provide with real data inside it?

meee1, 2016-04-02:
[image]

rmackay9, 2016-04-02:
that is really cool

meee1, 2016-04-02:
just working on the right balance of what to display, and what not to display.
as if I draw the projected footprint for every camera, MP slows down.... a lot.

meee1, 2016-04-02:
[image]

meee1, 2016-04-02:
new beta out.

jmachuca77, 2016-04-02:
This is awesome! is this working ?

iskess, 2016-04-02:
This is wonderful Michael! Thank you. 
I have a couple follow up requests:

1) It would be very nice to know if CAM_MIN_INTERVAL is being exceeded. Could you perhaps change the color of the camera icon to show that camera has hit its limit? In other words, if the distance between triggers is greater than CAM_TRIGG_DIST, then CAM_MIN_INTERVAL is delaying the triggering. This means that the overlap is being reduced which is very useful to know. 

2) Could we place a sequential number inside each camera icon so we can a) see how many shots have been taken, b) help us unravel geotagging mysteries by playing back the tlog and being able to more easily figure out which photo is missing. 

What are the purple polygons in the screen shot?

meee1, 2016-04-02:
[image]

1 - I will have to look into.

2 - the issue with numbers is there size, and if there are a large number of photos in the mission. I will need to look at the performance side, it is possible, just at what cost I need to determine.

the purple squares are the estimated ground footprint, based on camera RPY and terrain altitude

meee1, 2016-04-02:
to see the squares try the gridv2 with footprints turned on. this is a recent change, that may have gone unnoticed by many

iskess, 2016-04-03:
When I use the Survey Grid tool, and click on the Footprints, I see the shaded rectangles on the grid map just like always. Are you saying that this will activate a function that will allow me to see real time image footprints on the Flight Data map while flying? 
If so this would be fantastic! Is there another way to activate this function without needing to remember to do it when first creating the grid?
It appears from your screen shot that only the last five or so footprints are visible at a time to keep MP from slowing down too much. Is there a way to generate a map after the flight that shows all the footprints? This would be more like generating a graphic report from the flight log.

meee1, 2016-04-04:
[image]

this image shows what I mean.
the center of the grids ground elevation is high, so the footprint size is showing that

meee1, 2016-04-04:
if you mouse over points, it will show the footprint.

iskess, 2016-04-04:
Micheal, I like where this is going, but in reality its nearly impossible to visualize the individual footprints with the typical amount of overlap and sidelap. There must be a better way to present this information. What if we highlighted the just sidelap area and left out the rest of the footprint? Or somehow only display the areas where the sidelap drops below the target? What if we created a new field for minimum sidelap, and if the sidelap is calculated to drop below this threshold, then it is highlighted. For example, in mountainous terrain I choose a target sidelap of 75%, but I know I can live with as little as 50%. Then any areas with less than 50% are highlighted. I can then raise my altitude until all highlighting is gone. 
This would be even better if we could see it on the main Flight Plan map, not just the Survey Tool. This was we can change individual WP alts until the highlighting disappears. 
Ultimately we need a 3D survey tool, but I know I'm asking for too much there. 

How do you get the purple footprints to show up on the Flight Data map?

onurboy, 2016-04-04:
Micheal, thank you for your huge efforts bringing these desires to life. I am totally agree with the ideas of Iskess. If it increases the computational complexity instead of camera icons you can put only small dots with red,yellow and green color. We can understand that red is late, yellow is early and green is just in time taken photo. 
Instead of showing the numbers for all photos you can show only the latest one and we can always know the recent total amount of photos taken.
Thank you again.

Onur...

meee1, 2016-04-04:
the camera icon's are not the complexity, the footprints are, as they terrain intersect.

flightdata only displays after MAVLINK_MSG_ID_CAMERA_FEEDBACK message has been received.

meee1, 2016-04-04:
ive added green icons when the images are above CAM_MIN_INTERVAL, and red if they are bellow CAM_MIN_INTERVAL

iskess, 2016-04-04:
Michael,
I have a critical mission tomorrow. Is this available in the downloadable beta version yet? Do you think the beta is safe to use? In other words, were there areas changed that impact the basic functionality of MP? Hopefully I will have a good tlog for you to look at. 
Thanks.

meee1, 2016-04-04:
yes, this is all on current beta.
and yes safe to use, as its display only.

iskess, 2016-04-05:
beta is giving me an error and disconnects while loading parameters.
[image]

iskess, 2016-04-05:
When playing tlog in latest beta I can see trigger camera icons while the plane flies the flight lines. If I pause the playback, then resume, the waypoints and flight lines vanish. It appears that when I press pause, the map reverts to Lat/Long 0/0, and when I restart it, the map re-centers with all overlays removed.

meee1, 2016-04-05:
do you have a negative loiter radius?

iskess, 2016-04-05:
Yes, it's loiter radius is negative for left turns. I will update and try again. Thank you.

iskess, 2016-04-05:
still not working...
[image]

iskess, 2016-04-05:
Here is my param file if that helps:
[RMH with CAM changes and rollP 1.6to1.5.zip](https://github.com/ArduPilot/MissionPlanner/files/203763/RMH.with.CAM.changes.and.rollP.1.6to1.5.zip)

meee1, 2016-04-06:
i did fix the negative loiter issue, but forgot to post here. the workaround it just make it positive for the moment.

iskess, 2016-04-06:
Thanks Micheal, I did make the loiter positive to be able to fly the mission today. 
In all it went well. The camera icons really help to provide awareness of what the camera is doing. It is particularly helpful to see the camera stop and start outside the polygon (breakup starts). There were many missing icons however, which caused us some concern over whether we were triggering correctly. However the .bin shows that all photos were taken correctly, so the missing camera icons were just lost packets I suppose. 
During the flight, all the icons were red which is supposed to mean that CAM_MIN_INTERVAL was being invoked. However upon reviewing the .bin, none of the shots hit the min interval. When I play back the .tlog, all the icons are green, strange.
Could you change the color of the purple footprint so its not the same as the plane's track? Maybe a thinner line would do. 

Thank you!

meee1, 2016-04-06:
could you provide your tlog?

iskess, 2016-04-06:
https://www.dropbox.com/s/cp8xowvdb01gkhe/2016-04-05%2010-18-39%20quarry.tlog?dl=0

meee1, 2016-04-06:
the CAM_MIN_INTERVAL is fixed as part of #1216

Jman841, 2016-04-09:
Excellent work! I will test this shortly but I really like the improvements for mapping and visualization of the cameras!

meee1, 2016-04-13:
100% experimental
[image]

meee1, 2016-04-13:
[image]

iskess, 2016-04-13:
Very nice. That makes the coverage MUCH easier to visualize. However the map detail is total obscured. There would need to be a way to toggle this on/off. A color legend would also be helpful. For example, Dark Blue = 1 image, Green = 4 images, Red = 6 images, etc

meee1, 2016-04-13:
yes it does need a legend, and I'm very aware of the obscured, as I said, its experimental.

IronDome2, 2016-04-14:
Hi

Is the Auto-grid option removed? See attached.
How do I use/display the SRTM data?
[image]

Thanks in advance

meee1, 2016-04-14:
@irondome2 this is not the place for questions like that. however to fix it try updating again, your install is likely missing some files.

IronDome2, 2016-04-14:
My apologies, will look into thanks.

onurboy, 2016-04-18:
Mike,
Enhancement in comment [https://github.com/ArduPilot/MissionPlanner/issues/1210#issuecomment-209381835](url) slow downs the mission planner. I tried it with 3 different computers but all were the same. Could you check it please?

Thank you.

Onur.

meee1, 2016-04-18:
@onurboy yes it slows it down, depending on the coverage area checking 3000+ point on ever render does cause a large slow down. I am aware and will be addressing.

IronDome2, 2016-04-25:
I am not sure how this used.
But what about foot print polygon? Something similar to what is used in emotion
see this video as seem from 1:35
https://www.youtube.com/watch?v=BKzJdF_fk0U

Thanks

meee1, 2016-04-25:
ive seen it, and it doesn't show overlap, so don't like that implementation.

IronDome2, 2016-04-25:
I don't know for others, but most of the time I found overlap problems when doing missions that involve more than one flight. Most of the time it is hard to tell where the last flight ended. May be because I have not tried using segment splitting, usually if I many missions, I just create each mission separately. Each starting just at the end of the previous, hence the polygon footprint ensures that I am not leaving any gap between missions.

Thanks again

AquilaUAS, 2016-05-10:
I like the enhance in the latest beta. Is this safe to use yet?

closed 2017-04-20 by iskess

ACPUK, 2017-04-22:
Hi Iskess,
I used to until recently revived an icon in mission planner flight data map, of an image when the camera was triggered. I assume this is part of your development as above, However now this does not happen. Also I have hunted for the grids with footprints option within mission planner but don't see it.
"to see the squares try the grid with footprints turned on"

It this the 'Camera Overlay' under right click on map? if so it is not working for me,

thanks

iskess, 2017-04-23:
It works for me using MP 1.3.46 while playing back a tlog. 
The Camera Overlap displays the color dots that represent how many images include that area. The image footprints are automatic, but only a limited number of them persist on the display. By holding your pointer over an image icon, it will display the footprint from that image which is based on AGL altitude and airframe attitude.