ArduPilot/MissionPlanner #1212: Cannot resize info panel on flight data screen (1.3.37)

rsilk1949, 2016-04-01:
After I dragged the bottom edge of the HUD panel down to make it longer , I cannot resize it again.

meee1, 2016-04-01:
your not meant to be able to.
you can drag it to the left/right, but down/up is constrained by the left/right ratio

rsilk1949, 2016-04-02:
I can drag down, but not up. This behaviour has changed since 1.3.35, no issue with that release.
You can see the resize icon and drag it upwards but the display gets corrupted as you drag and when you release it snaps back down.
The HUD panel now takes 90% of the vertical space on the left of the map, and the info panel 10% and there is no way to change that.
Changing the overall window size or width of the map does not alter the vertical dimension of the HUD/info panels.

closed 2016-04-02 by rsilk1949