ArduPilot/MissionPlanner #1213: The simpleexample question in MP 1.3.37 source code !

goodman110110110, 2016-04-02:
When I build the simpleexample code to connect(click connect button) APM which has download the APM:Copter 3.3.3 firmware of copter, the programme can only receive the heartbeat packet and cann't receive the the attitude packet.
But I try to connect the APM which has download the ArduRover 2.5 or the ArduPlane firmware 3.5.2, the programme can receive the heartbeat packet and the attitude packet.
Please help me, Thankyou!

meee1, 2016-04-02:
please review the changes I made this morning.

closed 2016-04-02 by meee1

goodman110110110, 2016-04-02:
I just tried the newest changed simpleexample code, it does not work too.

meee1, 2016-04-02:
try connecting with mission planner first, then try the example. does that work?