ArduPilot/MissionPlanner #1214: Problems with Auto WP / Survey (Grid)

Seidenschwang, 2016-04-03:
Hi there,

I'm trying to create a mission for my hexacopter based on a polygon. Therefore I'm using the function "Auto WP" -> "Survey (Grid)".
[image]

But the following features doesn't work :-(:

- **"Grid Options"->"Copter Options"**: "Delay at WP" doesn't have any effects
[image]

- **"Camera Config"->"Trigger Method"**: DO_DIGICAM_CONTROL doesn't use the ShutterCommand: 
[image]

- **"Grid Options"->"Copter Options":** Heading Hold doesn't affect the footprints (Mission seems to be allright (Condition yaw) )
[image]

Do you have any idea about solving the problems?

(MissionPlanner: 1.3.37 / ArduCopter: 3.3.3-RC2)

meee1, 2016-04-04:
Thanks for the bug report. its much easy when you give me all the details, which you did a good job at!!

ive fixed all the issues you mention here.

meee1, 2016-04-04:
ive uploaded a new beta mp with the fixes

closed 2016-04-04 by meee1

Seidenschwang, 2016-04-04:
Wow, that was pretty fast!

Thank you