ArduPilot/MissionPlanner #1222: MAVLINK Connection Announcement

dtjw315, 2016-04-09:
I posted this on DIYDrones and Chris Anderson suggested I posted it here.

Ok, so I love Mission Planner. Also love the voice announcements and want to keep them...BUT.....when the mavlink first connects, it gives an announcement that is a series of numbers - "zero zero three to three zero zero three million three hundred...."  you get the idea - and takes about 3 and a half days to announce ha ha ha.

I know you can turn some announcements on and off in the config, but, is there anyway to turn that particular announcement off, or change what she says at that particular point?

It would be nice, as suggested above, to be able to disable or change this particular announcement.

meee1, 2016-04-09:
what version of copter are you running?

meee1, 2016-04-09:
ive added a permanent exclude for that one message.

iskess, 2016-04-09:
How about the message "Executing NAV Command #16" that pops up at every waypoint. Could that be in plain English?

meee1, 2016-04-09:
is it said? or just displayed?

iskess, 2016-04-09:
Both displayed and spoken. 
It also needs to know how to speak some abbreviations. For example a message says "past waypoint 56 dist 18m" but it spells out "d-i-s-t" rather than saying "distance" and says "one-eight-m" instead of "eighteen meters". It's very hard to decipher what it is saying.

meee1, 2016-04-09:
most of this is from the autopilot itself. Mp simply speaks what is passed to it.

iskess, 2016-04-09:
Is there a way to edit the speech engine to pronounce NAV and DIST, etc rather than spelling out each letter? 

What does "NAV Command ID #16" mean?

meee1, 2016-04-09:
I will have a look into that. this really should be a new issue however.

dtjw315, 2016-04-09:
I'm using v3.3.3

So, I'm new to copter and mission planner, so the code above, do I need to add it somewhere or just update my firmware?

iskess, 2016-04-09:
Go to the HELP tab and download the latest beta