ArduPilot/MissionPlanner #1223: Update Speech Engine so messages are understandable

iskess, 2016-04-09:
The current speech engine doesn't know how to speak some abbreviations and tends to read out digits individually. For example a message says "past waypoint 56 dist 18m" but it announces, "past waypoint five six d-i-s-t one eight m" the engine should be taught that "dist" is pronounced distance, 18 is eighteen, and m is meters. 
It's very hard to decipher what it is saying.

There is also a message i keep getting that has no meaning to me. "Executing NAV Command ID#16" it would be nice to have that in plain words. I think it's related to DO_SET_CAM_TRIGG_DIST, but not sure. 

NAV should not be pronounced N-A-V.

magicrub, 2016-04-10:
This came up at the last dev call, we wanted to make the text more human
readable but decided agreed it is definitely a GCS problem to solve. So,
FYI, this message format may be changing a little.

meee1, 2016-04-10:
my first take is I will do some word substitution.
ie
"dist" to "distance"
"nav" to "navigation"
"m" to "meters" (will need to be carefull with false positives on this one.)

iskess, 2016-04-10:
That's great!
It there a list of text messages that come down via mavlink so we can review other abbreviations that need associated words?

lvale, 2016-05-03:
on the source of ArduPilot search for gcs_send_text