ArduPilot/MissionPlanner #1224: Georeferencing Images Shutter Delay Problem

onurboy, 2016-04-10:
While using the georeferencing tool in mission planner if I enter any value in miliseconds to shutter lag section, the tool does not accept and does not geotag the pictures. I tried different values but did not help.

onurboy, 2016-04-13:
any idea???

meee1, 2016-04-13:
what error message do you receive? is it a . vs , issue?

onurboy, 2016-04-14:
Hi Mike,
I added the scrren capture. It says try a lower shutter lag number. I even tried one digit value but the same error occured.
[image]
Have u any idea?
Thanks.

Onur.

meee1, 2016-04-14:
your shutter lag value is to large.
here is the basic problem.

with a lag of 1.3 seconds and a speed of 15m/s  the distance error would be over 15m.
mp fails because the camera message time has varied to far away from the photo time (including the lag)

I would suggest processing the log using time offset vs CAM messages.
and also improving the shutter lag time. 1.3 seconds allows for way to much error

onurboy, 2016-04-14:
Mike,
Thank you for the answer but 1300 was just an example. I also tried smaller numbers like 100ms or 200ms but the results were the same. I think shutter delay feature does not work properly.

By the way do you know how I can exactly measure my shutter delay value?

Thanks.
Onur.

JacksonUAS, 2016-07-08:
Have exactly the same issue, any number in the shutterlag box results in errors (Time not found in log. Too Large Shutter Lag? Try setting it to 0). The camera we are using averages 282ms lag :/ Any solutions?

Patribu, 2017-06-20:
Hello, 
is there some solution for this issue?
I really want to avoid the offset method. CAM message is such a simple way to georef. the photos, if only the lag would work. For every number different from 0 I get errors. Does not matter if 1, 10 or 100, nothing works.

I would be very happy to hear from you.

Cheers