ArduPilot/MissionPlanner #1225: Spiralling Ascent / Descent Feature for Take-off & Landing

onurboy, 2016-04-10:
While planning a mission, it would be beneficial to be able to create auto spiralling climbing after take off and ascending before landing commands especially in rugged areas. Aircraft can gain altitude right after taking off above the home and vice versa for landing. 
I think it can be possible to implement this feature by automatically adding a group of commands in one button (right click menu etc.) on planning screen.
Thanks

Onur.

meee1, 2016-04-10:
can you provide a sample mission file of what it is you are expecting? this way I can visualise what you are asking.

onurboy, 2016-04-11:
Michael,
I will provide a sample mission file which defines what I expect, ASAP.
Thank you.

Onur.

onurboy, 2016-04-11:
Hi Michael,
I tried to define the waypoints.
You can find attached the files.
Thank you.

Onur.

magicrub, 2016-04-11:
@onurboy This would be nice to discuss at the dev call in 2 hours, are you able to get that example mission file posted before then?

magicrub, 2016-04-11:
woah, we posted at the same time :) I don't see any attachments @onurboy

onurboy, 2016-04-11:
I am just trying but could not upload the zip file

onurboy, 2016-04-11:
I made the extensions txt. You can change them to orginal waypoints extension
[spiral landing.txt](https://github.com/ArduPilot/MissionPlanner/files/213875/spiral.landing.txt)
[spiral take off.txt](https://github.com/ArduPilot/MissionPlanner/files/213874/spiral.take.off.txt)

magicrub, 2016-04-11:
Well that's not what I was expecting... Have you seen the LOITER_TO_ALT mission item?

onurboy, 2016-04-11:
Yes I know that item but i think I could not express myself. 
Maybe this video can help.
https://www.youtube.com/watch?v=AyPQ6j9prP8
Please check 1:51

magicrub, 2016-04-11:
yup, that's performing a loiter_to_alt.

onurboy, 2016-04-11:
do you think that loiter to alt command is sensitive and smooth as its in the video ? If so then I can use it instead.
Thank you.

Onur