ArduPilot/MissionPlanner #1226: cs code file missed.

JohnAizb, 2016-04-13:
The latest version:

Missed CSharp file: MissionPlanner\ExtLibs\Mavlink\MAVLink.cs.
CSharp Project file: MissionPlanner\ExtLibs\Mavlink.csproj
```
<Project .....>
  <ItemGroup>
    <Compile Include="**# Mavlink.cs**">
      <SubType>Code</SubType>
    </Compile>
......
  </ItemGroup>
......
</Project>

```
So: **MAV_PARAM_TYPE**  in MAVLinkPara.cs or other cs files are not defined anymore.
```

        public MAV_PARAM_TYPE Type { get; set; }

        private MAV_PARAM_TYPE _typeap = MAV_PARAM_TYPE.ENUM_END;
        public MAV_PARAM_TYPE TypeAP {
            get 
            { 
                if (_typeap != MAV_PARAM_TYPE.ENUM_END) 
                    return _typeap;
                return Type;
            }
            set
            {
                _typeap = value;
                
            }
        }

```

meee1, 2016-04-13:
mavlink.cs is autogenerated as part of the pre-build events.
please install python.

closed 2016-04-13 by meee1

JohnAizb, 2016-04-13:
Failed to generate the Mavlink.cs
```

1>来自项目“D:\GIT\MissionPlanner\ExtLibs\Mavlink\MAVLink.csproj”的文件“C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets”中的目标“PrepareForBuild: (目标 ID: 16)”(目标“CoreBuild”依赖于它):
1>正在使用程序集“Microsoft.Build.Tasks.Core, Version=14.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a”中的“FindAppConfigFile”任务。
1>任务“FindAppConfigFile” (任务 ID: 14)
1>  任务参数:
1>      PrimaryList=
1>          pymavlink\generator\lib\minixsv\datatypes.xsd
1>                  SubType=Designer
1>          pymavlink\generator\lib\minixsv\XInclude.xsd
1>                  SubType=Designer
1>          pymavlink\generator\lib\minixsv\xml.xsd
1>                  SubType=Designer
1>          pymavlink\generator\lib\minixsv\XMLSchema-instance.xsd
1>                  SubType=Designer
1>          pymavlink\generator\lib\minixsv\XMLSchema.xsd
1>                  SubType=Designer
1>          pymavlink\generator\mavschema.xsd
1>                  SubType=Designer (任务 ID: 14)
1>  任务参数:
1>      SecondaryList=
1>          message_definitions\ardupilotmega.xml
1>          message_definitions\common.xml
1>          pymavlink\generator\lib\genxmlif\xmlif4Dom.py
1>          pymavlink\generator\lib\genxmlif\xmlifApi.py
1>          pymavlink\generator\lib\genxmlif\xmlifBase.py
1>          pymavlink\generator\lib\genxmlif\xmlifDom.py
1>          pymavlink\generator\lib\genxmlif\xmlifElementTree.py
1>          pymavlink\generator\lib\genxmlif\xmlifMinidom.py
1>          pymavlink\generator\lib\genxmlif\xmlifODict.py
1>          pymavlink\generator\lib\genxmlif\xmliftest.py
1>          pymavlink\generator\lib\genxmlif\xmlifUtils.py
1>          pymavlink\generator\lib\genxmlif\__init__.py
1>          pymavlink\generator\lib\minixsv\minixsvWrapper.py
1>          pymavlink\generator\lib\minixsv\pyxsval.py
1>          pymavlink\generator\lib\minixsv\xsvalBase.py
1>          pymavlink\generator\lib\minixsv\xsvalErrorHandler.py
1>          pymavlink\generator\lib\minixsv\xsvalSchema.py
1>          pymavlink\generator\lib\minixsv\xsvalSimpleTypes.py
1>          pymavlink\generator\lib\minixsv\xsvalUtils.py
1>          pymavlink\generator\lib\minixsv\xsvalXmlIf.py
1>          pymavlink\generator\lib\minixsv\__init__.py
1>          pymavlink\generator\lib\__init__.py
1>          pymavlink\generator\mavcrc.py
1>          pymavlink\generator\mavgen.py
1>          pymavlink\generator\mavgen_csharp.py
1>          **pymavlink\generator\mavparse.py**
1>          pymavlink\generator\mavtemplate.py
1>          pymavlink\generator\__init__.py
1>          pymavlink\tools\mavgen.py
1>          pymavlink\tools\__init__.py
1>          pymavlink\__init__.py (任务 ID: 14)
1>  任务参数:TargetPath=MAVLink.dll.config (任务 ID: 14)
1>  输出项: AppConfigWithTargetPath= (任务 ID: 14)
1>已完成执行任务“FindAppConfigFile”的操作。 (任务 ID: 14)
1>任务“MakeDir” (任务 ID: 15)
1>  任务参数:
1>      Directories=
1>          bin\Debug\
1>          obj\Debug\ (任务 ID: 15)

```

```
1>  任务参数:WorkingDirectory=bin\Debug\ (任务 ID: 16)
1>  任务参数:Command=rd /s /q "D:\GIT\MissionPlanner\ExtLibs\Mavlink\mavlink"
1>  cd "D:\GIT\MissionPlanner\ExtLibs\Mavlink\"
1>  python -m pymavlink.tools.mavgen --lang=csharp --wire-protocol=1.0 "D:\GIT\MissionPlanner\ExtLibs\Mavlink\message_definitions\ardupilotmega.xml"  
1>  copy /y "D:\GIT\MissionPlanner\ExtLibs\Mavlink\mavlink\ardupilotmega\mavlink.cs" "D:\GIT\MissionPlanner\ExtLibs\Mavlink\Mavlink.cs"
1>   (任务 ID: 16)
1>  rd /s /q "D:\GIT\MissionPlanner\ExtLibs\Mavlink\mavlink"
1>  cd "D:\GIT\MissionPlanner\ExtLibs\Mavlink\"
1>  python -m pymavlink.tools.mavgen --lang=csharp --wire-protocol=1.0 "D:\GIT\MissionPlanner\ExtLibs\Mavlink\message_definitions\ardupilotmega.xml"  
1>  copy /y "D:\GIT\MissionPlanner\ExtLibs\Mavlink\mavlink\ardupilotmega\mavlink.cs" "D:\GIT\MissionPlanner\ExtLibs\Mavlink\Mavlink.cs"
1>   (任务 ID: 16)
1>  Traceback (most recent call last): (任务 ID: 16)
1>    File "C:\Program Files (x86)\Python34\lib\runpy.py", line 170, in _run_module_as_main (任务 ID: 16)
1>      "__main__", mod_spec) (任务 ID: 16)
1>    File "C:\Program Files (x86)\Python34\lib\runpy.py", line 85, in _run_code (任务 ID: 16)
1>      exec(code, run_globals) (任务 ID: 16)
1>    File "D:\GIT\MissionPlanner\ExtLibs\Mavlink\pymavlink\tools\mavgen.py", line 10, in <module> (任务 ID: 16)
1>      from pymavlink.generator import mavgen (任务 ID: 16)
1>    File "D:\GIT\MissionPlanner\ExtLibs\Mavlink\pymavlink\generator\mavgen.py", line 11, in <module> (任务 ID: 16)
1>      import mavparse (任务 ID: 16)
1>  ImportError: No module named '**mavparse**' (任务 ID: 16)
```

meee1, 2016-04-13:
ive tested with python27, please try that

JohnAizb, 2016-04-13:
After added "**from .**" before "import mavparse" in "**mavgen.py**", it works now, I think it should be applied.
Thanks anyway.

JohnAizb, 2016-04-13:
The python on my computer is 34.