ArduPilot/MissionPlanner #1232: missing Mavlink.cs

thanhly06, 2016-04-22:
anybody can give me file Mavlink.cs
I can't build MP without this file
thanks for watching

meee1, 2016-04-22:
please install python on your machine. the file is generated during build.

closed 2016-04-22 by meee1

magicrub, 2016-04-22:
Michael, perhaps we need a more useful error message. This particular
"problem" comes up often.

meee1, 2016-04-22:
Agreed, I might switch back to manually build as needed, and include the file in commits.
the reasons for this are
1. prevent this issue
2. there is no requirement to rebuild from the same xml's on every build
3. because of the change every build, all dependancys on this dll also get rebuilt.
4. ability to track change to the autogenerated file.

so in effect this will decrease compile time.

magicrub, 2016-04-22:
And there I go trying to be helpful and just end up causing a huge pain. Sorry!