ArduPilot/MissionPlanner #1233: Beta cannot download data flash logs

ThorstenDP, 2016-04-22:
Hi Michael,

the current beta of Mission Planner has some problems downloading data flash logs (tested on Win7 and Win10).
I only have the German error message:
**Error in log 10: Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.**
It works with the latest official build.
Tested on a PixRacer.

Cheers,
Thorsten

meee1, 2016-04-22:
this should be fixed now. thanks for the report. there maybe more small bugs in the current beta.. as there have been some large changes under the hood.

closed 2016-04-22 by meee1

ThorstenDP, 2016-04-22:
Thanks!
I am setting a Rixracer Quad and together with flying AC master and the beta versions of Mission Planner it is a little tricky to determine the source of some problems.
However, regarding the Initial Setup, Config/Tuning and the Log Browser all looks good in the latest beta version of Mission Planner now :-)