ArduPilot/MissionPlanner #1235: Georef Bug, Picture file name issue

raptortrias, 2016-04-23:
It seems that during geo ref if the file name in the camera has reached the end ie DSC999 and then it starts from the beginning DSC001 during a flight the script assumes the first picture is DSC 001 and this causes a problem with correct geo-tagging.

 Would it be easy to fix this?

 Thanks

meee1, 2016-04-23:
I agree that mp does suffer from this, as all the photo names when sorted are expected to be chronological.
I don't know what the fix for this would be atm though.

raptortrias, 2016-04-23:
If python based, a dictionary could provide mapping of file name vs exif: time, The exif time could be used to order the files,

 would this work?

meee1, 2016-04-23:
where you using cam messages? or time offset.

time offset pulls the exif info, cam messages does not, as it assumes the correct/name to begin with

raptortrias, 2016-04-23:
No, have been using time offset and was causing the problem. To be honest have not managed to make cam messages to work, camera is trigger with do set cam trig dist

meee1, 2016-04-23:
this should now be fixed

raptortrias, 2016-04-23:
Legend!
thanks

closed 2016-04-23 by raptortrias