ArduPilot/MissionPlanner #1237: PX4 'bin to log file' fails with arduplane V3.5.1 and above

t413, 2016-04-28:
Since V3.5.1 (V3.5.0 worked fine) using the mission planner bin-to-log conversion has failed silently and created zero-length files.

Here's a failing log file:
[apm-3.5.3px4log-21.BIN.zip](https://github.com/ArduPilot/MissionPlanner/files/241081/apm-3.5.3px4log-21.BIN.zip)

(Logs also fail to read in the latest APM Planner 2)

Where can I find documentation on this file format? It would be good to verify whether this is a bug in ArduPlane's log creation or in MissionPlanner's log reading.

ctonyperry, 2016-04-28:
I am using the same version of Arduplane and getting similar results. When I view the file in a hex editor it appears to be 18mb of zeros.

Here's a copy of my file.
[apm-3.5.3px4log-140.BIN.zip](https://github.com/ArduPilot/MissionPlanner/files/241105/apm-3.5.3px4log-140.BIN.zip)

t413, 2016-04-28:
Ahh. Because it's NULL-extended like that I'm opening an issue on ardupilot: https://github.com/ArduPilot/ardupilot/issues/4009

t413, 2016-04-30:
Closed over at ArduPilot/ardupilot#4009, it's actually the fault of a scam SD card we both had. [How to check and spot fake Micro SD card](http://www.happybison.com/reviews/how-to-check-and-spot-fake-micro-sd-card-8/)

closed 2016-04-30 by t413