ArduPilot/MissionPlanner #1243: Added option to Mission Planner to have output turned on only when in survey area

jagrandle, 2016-05-05:
I understand that a camera can be triggered at distance intervals when only in the survey area using the Breakup Starts option.

It would be great if a similar type of output or servo position could be turned ON only when in the Survey area, and OFF when outside it.

Thanks fro your help

jagrandle, 2016-06-03:
Hi Michael,
Its been around a month since I put up this suggestion or request. Is there any way for me to know if you have actually seen it ?
I understand there may well be much higher priorities for you to work on, but it would be re-assuring to know if you have seen this or if there is any chance of this option being implemented.
Thanks

meee1, 2016-06-03:
can you provide a sample mission?
just want to see your use case so I can address it.

jagrandle, 2016-06-03:
Wow, that is a fast reply, thanks !!

Either fixed wing or multi.
Have an item which I only want to be turned on whilst in survey area, 
and off when outside survey area during transits or for overshoots 
during turnaround

Typical items to be switched  eg/ light, motor, dispenser, trap door 
open, transmitter, smoke generator or whatever.
Probably easiest if this feature just controls a servo output like the 
normal CAM_TRIGG-DIST etc

I know that with the new option "Breakup Starts" the camera will only 
trigger at the required distances within the Survey Area it would seem 
that there is software checking when the aircraft is inside/outside the 
area.

Maybe it could be a fourth option under               Camera 
Config        Trigger Method
and allow the user to select the           Servo #         PWM whilst 
inside survey Area          PWM whilst outside

Open to any other options or suggestions you might have.

Thanks very much for getting back to me,
Jeremy

Senior Research Engineer
Australian Centre for Field Robotics
University of Sydney
Australia
Mob: 0418 390 446
Ph:  61-2-9351-7468
Fax: 61-2-9036 7474

magicrub, 2016-06-03:
How about a sample waypoint file?

jagrandle, 2016-06-04:
Hi Tom,

I have Replied to All but if this is not desired, please let me know 
what I should use in future.

Attached is both a Waypoint file and the Grid file.

Essentially I have something which I only wish to be activated whilst in 
the Survey Area. Instead of having:

DO_SET_CAM_TRIGG_DISTANCE     which sets a value of x metres at the 
start of each leg when entering the survey boundary and then sets a 
value of 0 metres when exiting the survey area

I was hoping for something like:

DO_SERVO_ON    and     DO_SERVO_OFF        where one can specify the 
Servo number and PWM value for ON and OFF and these commands were 
inserted automatically in the flight plan at the entry and exit of the 
servo area.

I am sure there are other ways around it, this is just a suggestion to 
try and convey my meaning.

Thanks for your time,
Jeremy

Senior Research Engineer
Australian Centre for Field Robotics
University of Sydney
Australia
Mob: 0418 390 446
Ph:  61-2-9351-7468
Fax: 61-2-9036 7474

QGC WPL 110
0	1	0	16	0	0	0	0	-34.594577	150.054435	620.000000	1
1	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595620	150.051024	50.000000	1
2	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595810	150.051394	50.000000	1
3	0	3	206	24.375000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
4	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.597485	150.054659	50.000000	1
5	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
6	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.597675	150.055029	50.000000	1
7	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.597275	150.055312	50.000000	1
8	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.597085	150.054942	50.000000	1
9	0	3	206	24.375000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
10	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595419	150.051694	50.000000	1
11	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
12	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595229	150.051324	50.000000	1
13	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.594838	150.051624	50.000000	1
14	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595028	150.051995	50.000000	1
15	0	3	206	24.375000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
16	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596685	150.055225	50.000000	1
17	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
18	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596875	150.055595	50.000000	1
19	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596475	150.055879	50.000000	1
20	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596285	150.055508	50.000000	1
21	0	3	206	24.375000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
22	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.594637	150.052295	50.000000	1
23	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
24	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.594447	150.051925	50.000000	1
25	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.594056	150.052225	50.000000	1
26	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.594246	150.052595	50.000000	1
27	0	3	206	24.375000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
28	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595885	150.055791	50.000000	1
29	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
30	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596075	150.056162	50.000000	1
31	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596675	150.055737	50.000000	1
32	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596485	150.055367	50.000000	1
33	0	3	206	24.375000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
34	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.594832	150.052145	50.000000	1
35	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
36	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.594642	150.051775	50.000000	1
37	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.594251	150.052075	50.000000	1
38	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.594441	150.052445	50.000000	1
39	0	3	206	24.375000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
40	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596085	150.055650	50.000000	1
41	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
42	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596275	150.056020	50.000000	1
43	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.597075	150.055454	50.000000	1
44	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.596885	150.055083	50.000000	1
45	0	3	206	24.375000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
46	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595223	150.051845	50.000000	1
47	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
48	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595033	150.051474	50.000000	1
49	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595424	150.051174	50.000000	1
50	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.595614	150.051544	50.000000	1
51	0	3	206	24.375000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
52	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.597285	150.054800	50.000000	1
53	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
54	0	3	16	0.000000	0.000000	0.000000	0.000000	-34.597475	150.055171	50.000000	1
55	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1

meee1, 2016-06-04:
would the more simple solution be for me to set the do_repeat_servo repeat count to 1? from 999.

this means the servo would only actuate once per internal WP. and then stop.

jagrandle, 2016-06-04:
I don't think I want to have any internal WP.

I wish to have it trigger ON when entering the Survey Area and stay ON 
continuously until it leaves the Survey Area, at which time it turns OFF.

Thanks

Senior Research Engineer
Australian Centre for Field Robotics
University of Sydney
Australia
Mob: 0418 390 446
Ph:  61-2-9351-7468
Fax: 61-2-9036 7474

meee1, 2016-06-04:
ok here a sample wp file I just created

QGC WPL 110
0	1	0	16	0	0	0	0	-34.980762	117.851967	36.000000	1
1	0	3	22	20.000000	0.000000	0.000000	0.000000	0.000000	0.000000	30.000000	1
2	0	3	178	0.000000	12.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
3	0	3	16	4.000000	0.000000	0.000000	0.000000	-34.979287	117.850810	100.000000	1
4	0	3	184	5.000000	1100.000000	999.000000	2.000000	0.000000	0.000000	0.000000	1
5	0	3	16	4.000000	0.000000	0.000000	0.000000	-34.982532	117.851218	100.000000	1
6	0	3	184	5.000000	1100.000000	0.000000	2.000000	0.000000	0.000000	0.000000	1
7	0	3	16	4.000000	0.000000	0.000000	0.000000	-34.982604	117.852143	100.000000	1
8	0	3	184	5.000000	1100.000000	999.000000	2.000000	0.000000	0.000000	0.000000	1
9	0	3	16	4.000000	0.000000	0.000000	0.000000	-34.979329	117.851732	100.000000	1
10	0	3	184	5.000000	1100.000000	0.000000	2.000000	0.000000	0.000000	0.000000	1
11	0	3	16	4.000000	0.000000	0.000000	0.000000	-34.979371	117.852653	100.000000	1
12	0	3	184	5.000000	1100.000000	999.000000	2.000000	0.000000	0.000000	0.000000	1
13	0	3	16	4.000000	0.000000	0.000000	0.000000	-34.982675	117.853068	100.000000	1
14	0	3	184	5.000000	1100.000000	0.000000	2.000000	0.000000	0.000000	0.000000	1
15	0	3	16	4.000000	0.000000	0.000000	0.000000	-34.982746	117.853993	100.000000	1
16	0	3	184	5.000000	1100.000000	999.000000	2.000000	0.000000	0.000000	0.000000	1
17	0	3	16	4.000000	0.000000	0.000000	0.000000	-34.979414	117.853574	100.000000	1
18	0	3	184	5.000000	1100.000000	0.000000	2.000000	0.000000	0.000000	0.000000	1
19	0	3	20	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1

jagrandle, 2016-06-04:
Hi,

So I cut and pasted the sample test in to a waypoint file. Afraid I 
can't read WP by human yet !!

What I see is a lawn mower flight profile. There is no Polygon area (ie/ 
Survey Area) so I can not tell which portion of the flight path is 
inside or outside of it.

At every point the DO_REPEAT_SERVO command is triggered with a PWM of 
1100 and cycle time of 2sec for 999 repeats. This means that there are 
multiple actuations of the servo. I would like it to just move once to 
on when entering the area, then to off when leaving.

I have attached a Photoshoped image of how I imagine a Flight Plan might 
look. I have tried to indicate where the Servo might turn on or off. 
Hopefully it helps

There already is the DO_SET_SERVO command. I am just after an automated 
way of having this inserted in a flight plan with a value just like 
setting the camera trigger distance is set to xx metres on the way in to 
the Survey Area and to metres on the way out.

Senior Research Engineer
Australian Centre for Field Robotics
University of Sydney
Australia
Mob: 0418 390 446
Ph:  61-2-9351-7468
Fax: 61-2-9036 7474

meee1, 2016-06-04:
the grid interface was designed to take photos during a mission. it sounds like your use for it 100% custom.

I have to ask, what is the intended use?

jagrandle, 2016-06-04:
Being a University we try many things, just ask Steve Keep who I used to 
work with ! I believe you know Steve :-)

We have potential applications to:
Turn on lights (eg/ indicate they are in a certain area)
Deploy smoke (eg/ another visual indication)
Turn on motors or pumps
Open hatches or covers (eg/ camera protection etc)
Turn on sounds (eg/ bird repellant)

I guess I thought it could be a fourth option under the   Camera 
Config   tab
I thought a lot of the logic and position checking used for the other 
options, such as breakup starts,  could be re-used.
Also, similar input boxes such as those for the    DO_REPEAT_SERVO 
command could also  be used.

Sorry if I a making things difficult. I do appreciate your help on a 
Saturday night, albeit a wet one here in Sydney. I have to run to the 
supermarket to buy food now so replies will be a little slower. I am 
still very interested if this is possible.

On another note, is there any way to change the column widths in the 
Flight Planner menu so you can see the whole command name or variable 
value ??

Thanks

Senior Research Engineer
Australian Centre for Field Robotics
University of Sydney
Australia
Mob: 0418 390 446
Ph:  61-2-9351-7468
Fax: 61-2-9036 7474

jagrandle, 2016-07-23:
I just upgraded to the latest version of Mission Planner 1.3.9 and it seems that this feature has been implemented. Thanks team !!

I made a quick flight plan with lead in and lead out areas. Sure enough it placed the appropriate servos commands in the flight plan at the start and end of every leg which was inside the area of interest. Great.

Now if only it were possible to resize the widths of the Flight Plan columns as one could actually read tall the numbers.....but I guess everybody has already requested that !!  ;)