ArduPilot/MissionPlanner #1246: Report Onboard compass failure

rmackay9, 2016-05-09:
In Copter-3.4 there's a new onboard compass calibrator.
When the compass calibration fails, there's no clear indication in mission planner that it's failed.
Here's a tlog, .bin and image showing what mission planner shows during a failed compass calibration.
[failed-compass-cal.zip](https://github.com/ArduPilot/MissionPlanner/files/254188/failed-compass-cal.zip)

rmackay9, 2016-05-09:
It also doesn't seem to show success..

meee1, 2016-05-09:
[image]
acording to the log both compass's reach 99 percent. so did it fail?

meee1, 2016-05-09:
after looking a bit more, the calibration finished successfully, even though the offsets the calibration detected where 0. not sure how that can happen.

WickedShell, 2016-10-22:
Not a MP issue. You can see the referenced ArduPilot PR for a initial patch that addresses it.

closed 2016-10-27 by meee1