ArduPilot/MissionPlanner #1247: Request: Changing the orientation of map in flight plan section

onurboy, 2016-05-09:
Mike,
I do not know if its already included but I cannot find changing the direction of the map (North-South-East-West etc.) and North always on the top in Flight Plan section.
I think it would be useful while planning in the field to be able to change the orientation of the map.

Thanks.

Onur.

onurboy, 2016-06-19:
@meee1 
Mike,
I know you are busy with other priorities but is that possible to handle with the issue above?
Thank you.

Onur.

iskess, 2016-06-19:
I had the chance to use MP in a manned aircraft to fly flight lines and use an APM to trigger a camera by distance. It worked pretty well except for the fact that MP doesn't have a moving map display. If it was possible to move the map around a stationary plane icon, it would be very usable for this purpose. As it is, it's much harder to hand fly the plane to stay on the flight line.  I realize this is not what MP was designed to do, but I thought I'd mention it.

taedla01, 2016-06-20:
Hello everyone!

That´s a great addition @meee1. Thanks for propose @iskess! ;)

Cheers,
Dario.