ArduPilot/MissionPlanner #1248: Failed mission upload can result in waypoints from previous mission remaining

AquilaUAS, 2016-05-09:
I experienced an issue last Thursday whilst doing mapping flights, the first 2 flights completed as normal however the third somehow managed to complete 121 waypoints out of a total 187 and then proceeded to head off in the direction of the previous mission. RTL was invoked and the plane brought home and landed. When looking at the BIN file the waypoints were those of the previous mission. I'm left confused as to what happened during flight 3. There were a couple of failed waypoint uploads on an earlier flight in the day and whether this is related or not I'm not sure. I have put the tlog and Bin's in a dropbox folder . The 2 logs to look at are 2016-05-05 14-39-35 & BIN442. The mission previous to this (the one that the plane reverted to after WP121) are 2016-05-05-13-03-31 & BIN441 

https://www.dropbox.com/sh/fm5wozoec9u2dz4/AAAADlixI6Pcsizno5SWkBDva?dl=0

This issue was started in https://github.com/ArduPilot/ardupilot/issues/3338

meee1, 2016-05-10:
do you have the tlog for part 3? i only see part 1 and 2 tlog's

meee1, 2016-05-10:
what version of mission planner where you running?

meee1, 2016-05-10:
fixed here
https://github.com/ArduPilot/MissionPlanner/commit/dfbce268dfc916ed5310306bd03b7abc366d3353

AquilaUAS, 2016-05-10:
I was using 1.3.37 Build 1.1.5917.13431. Tlog from the first flight to match BIN 439 is in the same dropbox folder (2016-05-05-10-54-55)

https://www.dropbox.com/sh/fm5wozoec9u2dz4/AAAADlixI6Pcsizno5SWkBDva?dl=0

Will updating MP sort this bug out?

meee1, 2016-05-10:
Thank you for the log. without the log i would not have been able to find this. the issue was a MP problem, to do with io lag. ie
mp sent a mission item, got no answer, so sent it again, then the autopilot responded to the first request and advanced the wp it was asking for, but the second packet mp sent was still in the buffer, and was responded to with an invalid sequence error, mp then assumed the invalid sequence was because it didn't see the packet response, and sent the next wp.
now it will confirm the next requested wp before proceeding.

AquilaUAS, 2016-05-10:
Ok thats good to know what happened, I can have confidence on my next outing (Thursday to continue with this mapping job) I installed the latest beta version of MP ( I like the camera postions and footprint indications) Is this version stable to use in the field?

meee1, 2016-05-10:
the camera positions and footprints are display only, so yes it should be safe, it makes no modifications to flight

AquilaUAS, 2016-05-10:
Nice one. This work is fantastic.

AquilaUAS, 2016-05-10:
http://www.dronelab.io/map/public/view/dde51fa7eba04efc87da4b5393de154a
This is part of the job I'm working on, this is the first 10km of 26km

AquilaUAS, 2016-05-10:
The issue happened on 13km-14km section

meee1, 2016-05-10:
nice job, I'm surprised you don't fly the long length. as you would increase you coverage area due to fewer turns

AquilaUAS, 2016-05-10:
I flew that direction due to the wind, it was up around 7-8m/s and I like to maintain a constant ground speed to prevent camera buffer overload. I had long length flights at the earlier stages of the job