ArduPilot/MissionPlanner #1249: MultiVehicle support - one vehicle missing from drop-down

stefanlynka, 2016-05-12:
While connecting the mission planner to an antenna tracker (which was in turn connected to a quadcopter), the vehicle selection drop-down only showed the tracker.  See pic below which shows both vehicles were visible on the map.
[image]

[tlog_tracker_and_quad.zip](https://github.com/ArduPilot/MissionPlanner/files/260577/tlog_tracker_and_quad.zip)

rmackay9, 2016-07-05:
I'd like to confirm that this issue is there.  I saw it today as well.

meee1, 2016-07-05:
update here
6e4a7d1f4fe8361633e20c8afe1912a190c000d3

rmackay9, 2016-07-05:
Thanks Michael!

meee1, 2016-07-05:
this change above is to just update the list.
it occurs because the seconds vehicle was not connected during mp connect.
ie tracker was on, but copter was not.

there may still be a further bug around parameters on the copter in this situation. more testing needed