ArduPilot/MissionPlanner #1257: lost communication with the board

chenfengfeng, 2016-05-21:
Upload Firmware Error with pix4-v2

meee1, 2016-05-21:
more information required.

chenfengfeng, 2016-05-21:
>System:windows 10 pro build 14332
Firmware version:Copter 3.3.3/3.4
MP version:1.3.8
pix4 COM:3

Firmware downloaded from the official website.

Loading firmware and then reconnect the USB tips:
>lost communication with the board
Error uploading firmware

meee1, 2016-05-21:
please try another pc, and a different usb port, and different usb cable.

proficnc, 2016-05-21:
You need good quality USB cables.  Also connect directly to your computer, avoid hubs.

chenfengfeng, 2016-05-21:
try to replace the USB and device,but same results

meee1, 2016-05-21:
please upload your missionplanner.log file from the mp install directory

chenfengfeng, 2016-05-21:
```
2016-05-21 16:19:32,016  INFO Program - ******************* Logging Configured ******************* (c:\Users\michael\Source\Repos\MissionPlanner\Program.cs:49) [1]
2016-05-21 16:19:32,188  INFO MissionPlanner.MainV2 - Mainv2 ctor (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:411) [Base Thread]
2016-05-21 16:19:32,188  INFO MissionPlanner.MainV2 - Loading config (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:1708) [Base Thread]
2016-05-21 16:19:32,250  INFO MissionPlanner.Utilities.Speech - TTS: init, mono = False (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\Speech.cs:43) [Base Thread]
2016-05-21 16:19:32,266  INFO MissionPlanner.MainV2 - change lang to en-US current zh-CN (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2814) [Base Thread]
2016-05-21 16:19:32,298  INFO MissionPlanner.MainV2 - Create FD (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:648) [Base Thread]
2016-05-21 16:19:32,313  INFO MissionPlanner.GCSViews.FlightData - Ctor Start (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:171) [Base Thread]
2016-05-21 16:19:32,642  INFO MissionPlanner.GCSViews.FlightData - Components Done (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:175) [Base Thread]
2016-05-21 16:19:32,642  INFO MissionPlanner.GCSViews.FlightData - Tunning Graph Settings (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:193) [Base Thread]
2016-05-21 16:19:32,657  INFO MissionPlanner.GCSViews.FlightData - HUD Settings (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:232) [Base Thread]
2016-05-21 16:19:32,657  INFO MissionPlanner.Common - getModesList Called (c:\Users\michael\Source\Repos\MissionPlanner\Common.cs:240) [Base Thread]
2016-05-21 16:19:32,672  INFO MissionPlanner.GCSViews.FlightData - Graph Setup (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:277) [Base Thread]
2016-05-21 16:19:32,672  INFO MissionPlanner.GCSViews.FlightData - Map Setup (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:281) [Base Thread]
2016-05-21 16:19:37,720  INFO MissionPlanner.MainV2 - Create FP (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:650) [Base Thread]
2016-05-21 16:19:37,782  INFO MissionPlanner.MainV2 - Create SIM (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:653) [Base Thread]
2016-05-21 16:19:37,813  INFO MissionPlanner.MainV2 - myview width 1008 height 1017 (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2673) [Base Thread]
2016-05-21 16:19:37,813  INFO MissionPlanner.MainV2 - this   width 1024 height 1056 (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2675) [Base Thread]
2016-05-21 16:19:37,813  INFO MissionPlanner.MainV2 - myview width 1920 height 1017 (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2673) [Base Thread]
2016-05-21 16:19:37,813  INFO MissionPlanner.MainV2 - this   width 1936 height 1056 (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2675) [Base Thread]
2016-05-21 16:19:37,813  INFO MissionPlanner.MainV2 - Saving config (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:1724) [Base Thread]
2016-05-21 16:19:37,876  INFO MissionPlanner.MainV2 - Load Pluggins (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2387) [Base Thread]
2016-05-21 16:19:37,907  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\MissionPlanner.Grid.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:75) [Base Thread]
2016-05-21 16:19:37,907  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init Grid 1.0.5983.12146 by Michael Oborne (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:86) [Base Thread]
2016-05-21 16:19:37,907  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\MissionPlanner.Gridv2.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:75) [Base Thread]
2016-05-21 16:19:37,907  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init Grid 0.1 by Michael Oborne (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:86) [Base Thread]
2016-05-21 16:19:37,922  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\MissionPlanner.SimpleGrid.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:75) [Base Thread]
2016-05-21 16:19:37,922  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init SimpleGrid 0.1 by Michael Oborne (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:86) [Base Thread]
2016-05-21 16:19:37,922  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\MissionPlanner.Stats.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:75) [Base Thread]
2016-05-21 16:19:37,954  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init Stats 0.1 by Michael Oborne (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:86) [Base Thread]
2016-05-21 16:19:37,954  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\TrackerHome.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:75) [Base Thread]
2016-05-21 16:19:37,954  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init GPS Tracker Home Plugin 1.0 by Will Bryan (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:86) [Base Thread]
[... cut ...]
   at MissionPlanner.Utilities.Firmware.UploadPX4(String filename, boards board) in c:\Users\michael\Source\Repos\MissionPlanner\Utilities\Firmware.cs:line 765 (c:\Users\michael\Source\Repos\MissionPlanner\Utilities\Firmware.cs:769) [Base Thread]
2016-05-21 16:21:03,948  INFO MissionPlanner.MainV2 - myview width 0 height 0 (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2673) [Base Thread]
2016-05-21 16:21:03,948  INFO MissionPlanner.MainV2 - this   width 160 height 28 (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2675) [Base Thread]
```

meee1, 2016-05-21:
its either bad power, or a bad usb cable.
the last error in the log shows that the process failed during a write.

chenfengfeng, 2016-05-21:
I found I could use QGC to upload firmware

meee1, 2016-05-21:
what are the specs on your machine. I think it might be because of a write timeout.

meee1, 2016-06-03:
I did make a change regarding this, the default timeout was set low, so I have increased the timeout period, and think will resolve this issue.

meee1, 2016-06-03:
changed here
https://github.com/ArduPilot/MissionPlanner/commit/918975b7b63b6e3ec3820397847d91343e52cd23

could you please test with the latest beta MP?

chenfengfeng, 2016-06-08:
Thank you very much, I try it