ArduPilot/MissionPlanner #1258: Sysid selector (ctrl-x) removed?

ggregory8, 2016-05-23:
Hi,

It looks like the ctrl-x shortcut has been removed to select which vehicle/unit (in a system with multiple sysid) we will display data and parameters for.

Has this been replaced by something else?

How do we select which sysid to use?

Thanks.

meee1, 2016-05-23:
it was replaced with the drop down up the top, the one under the baud rate.

iskess, 2016-05-24:
What replaced the hidden secret menu that forced MP into copter or plane mode? Sometimes I find myself in the wrong mode and I am not nearby an aircraft to connect to. For example, I'm trying to create a plane flight plane but I'm in copter mode and the commands are different.

meee1, 2016-05-24:
currently there is no secret.

try connecting to sitl first though, as that will make it change

meee1, 2016-06-03:
closing as functionality was moved.

closed 2016-06-03 by meee1