ArduPilot/MissionPlanner #1259: New grid option for survey

buckker, 2016-05-24:
I miss the option to program a simple grid of for example 10x10 meter. Mission Planner has already the option "distance between lines". But this works only for one direction (i call it X direction). Very nice would be if we have the same feature for the Y direction.

In the 1.3.35 version it's possible to get a grid of 10x10 Meter by adjusting the focal length and the distance between lines. With 1.3.38 the focal length do nothing, the grid won't be updated.

BTW, I use Win10

meee1, 2016-05-24:
in 1.3.38 there is an option for cross grid, that will draw a grid in both X and Y.
to adjust distance between lines empty the camera selection box. and you will be able to change the distance between lines.

buckker, 2016-05-25:
Hi Michael

Yes I have tested it. But I don't get a command like "do repeat servo" every 10 meter. I got a very strange pattern of waypoints:-)
[image]

Many thanks in advance

Michael

meee1, 2016-05-25:
Survey grid is designed to work with a camera. you are obviously not doing that. have you looked at using simplegrid? and defined the do set servo yourself?

buckker, 2016-05-27:
Hi Michael

With simple grid it works perfect :-) except the do set servo command.. How do define that? Manually after every waypoint? This would be a huge work for every mission.