ArduPilot/MissionPlanner #1262: Georeferencing: can not use CAM msgs from tlog files.

HernanBorcano, 2016-05-27:
I have been using time offset since I started with aerial mapping with many flights. Now to get a better workflow I tried to use cam msgs from tlog, and get this:  "Log file problem. No CAM messages. Aborting..." 
I know tlog has CAMERA_FEEDBACK msgs, so why it does not find them. Also When looking to photo coverage projection on flight data I see that it does a better job at placing images tan Georeferencing. One needs to fickle too much with dir fov and cross fov to get nice results when looking at generated kml on google earth, and even then it does not do the same proyection/warping that you see in flight data. To ilustrate what I´m talking about:

[image]

[image]

When using further processing with Photoscan for example This KML is not important but, as a quick check or even to make short flights with lower image count It is very useful. For many agricultural uses as crop monitoring for example you need resolution but mosaic generation is not needed and overkill, too much data. Submeter precision not needed also but a better general orientation of images would work very well.
I want to help on translation of MP to Spanish also, as a way to return something to ardupilot and MP, as I been enjoying them from more tan 2 years as a user/tinkerer. I´m new to github and almost 0 knowledge in programing meaning no more tan a Little arduino.

Cheers
Hernan

meee1, 2016-05-28:
is there any chance you could share your photoset and logs?

HernanBorcano, 2016-05-28:
I will try to upload it as son as posible.

Thankyou Michael

HernanBorcano, 2016-05-28:
https://www.dropbox.com/sh/zewdxsfwqfz8r9q/AABhVcVSnORoZ0DqYEhSwSdta?dl=0
Uploading now to Dropbox. Full flight log with pictures. Battery on camera died as thi was the last flight of the day so not all pictures were obtained. Camera is in landscape orientation in flgiht direction, but with top looking to tail of plane.

meee1, 2016-06-07:
first thing i notice, you gave me the tlog. the tlog does not contain cam messages. tlogs can only be used with the time offset method.

HernanBorcano, 2016-06-07:
Michael, then why I can graph cam feedback msgs with tlog? I know the best option is onboard logs but I think for many frames sdcard is not so easy accessed and log download is painfully slow. One problema with tlogs would be signal dropouts, but with good radios (RFD900) we get a few kilometers with just about 0 dropped packets. 
I hope I do not sound annoying with this request, thats not the idea. I think it would be for many a better choice tan onboard logs.
Cheers
Hernan
[image]

meee1, 2016-06-07:
the problem is that camera feedback messages can go missing, with no trace. and can causes offset issues.

HernanBorcano, 2016-06-07:
Ok so there should be at least a camera feedback msg like a counter, I Think I´ll go to annoy a little the ardupilot guys. 
I`m about to swap apm for pixhawk on the plane so I`ll try to do another example with onboard logs.

Thanks Michael!

HernanBorcano, 2016-06-07:
An off topic question. If I Try to do some more translating in spanish within MP lang edit option, How can I get it to you, or here?

meee1, 2016-06-07:
when I say no trace, I mean that the message is already index'ed however if it goes missing, MP has nothing to work from.

CAM message synchronisation in MP was never designed to work with tlogs. only DF Logs

also a pixhawk will solve your log download times.

meee1, 2016-06-07:
on here is best, as that way it is tracked. and I don't miss it.