ArduPilot/MissionPlanner #1265: Omega-Phi-Kappa values in Cam Messages

onurboy, 2016-05-27:
Mike,
Currently when we get geotagging info from the logs, omega, phi, kappa values in cam messages are with only 2 digits after the comma (ex: 12.21, -4.49, 250.17).

Is it possible to increase the number of digits after comma? I need at least 6 digits after comma to be able to get a better precision. I noticed that IMU messages have 7 digits after the comma in the log.

Thanks

Onur.

meee1, 2016-05-28:
I'm not limited that in any way, so the source data is only providing 2 decimal points.

are you using tlogs or dflogs?

onurboy, 2016-05-28:
I am using dflogs.
opened dflog in text editor and in CAM message lines values have only 2 decimals, but in IMU message lines values have 7 decimal points...

meee1, 2016-06-01:
i personally believe the precision you are chasing is unrealistic. and also the time tagging of the imu message vs the cam message would create a larger error than just using the value rounded to 2 decimal places.

onurboy, 2016-06-19:
OK. It is just a user requirement who uses Stereoscopic CAD method for drone images 3D inspection and plotting. They asked me if it was possible or not to have 5 or 6 decimals to increase the precision.
I am not sure the precision is unrealistic or not.

Thank you.

Onur.