ArduPilot/MissionPlanner #1268: Extended Tuning screen

rmackay9, 2016-05-30:
Not sure we should have the "Y", "X" and "Z" attached to the end of the "Rate Roll", "Rate Pitch" and "Rate Yaw".
"Rate Roll Y" -> "Rate Roll"
"Rate Pitch X" -> "Rate Pitch"
"Rate Yaw Z" -> "Rate Yaw"

If I input a number that is out of range for Rate Roll P it pops up a warning (which is good), "ATC_RAT_RLL_P Value out of range, Do you want to accept the new value?".  If I push "No" it should revert the parameter to it's original value, not the maximum (or minimum) of the parameter's valid range.

tridge, 2016-05-30:
when I test it on windows8, also with latest MP beta, I don't get the error popup when an out of range value is entered in the extended tuning screen. The value just immediately reverts to the old value with a short beep. That confused a person tuning a vehicle with Leonard a lot.
Also, in the "standard params" screen there is an oddity where the max value is the current value if current value is above the max. Then if you lower the value a bit and write it re-scales so you can't take the value back again.

meee1, 2016-05-30:
@rmackay9 the check you hit is a simple check of if the value has more than doubled. its not a range issue. just a fact that the value has more than doubled from the previous.
if ((float) changes[value] > (float) MainV2.comPort.MAV.param[value]*2.0f)

@tridge the extended tuning screen is limited to the min's and max's defined in the code metadata. if it is outside the default range then yes the sliders are setup to extend to the current setpoint. this really comes down to are the default ranges correct?

When updating to beta MP's master is used for the metadata, when using stable MP the stable branch is used.

caveat. when params are renamed, depending on your metadata version (affects beta mp) then there may not be a valid min/max range. so the max will be set to the current value, as its a safe option given no information.