ArduPilot/MissionPlanner #1269: Download Logs does not rename filenames according to GPS date since 1.3.38

t2000, 2016-05-30:
Hello,

since I updated MissionPlanner to version 1.3.38 I can still download my log files from my APM. However, the filenames are 1 1-1-1970 1-00-00 AM.bin and not like the GPS date+time within the log file. Before in version 1.3.37 MissionPlanner looked INTO the logfiles, extracted the GPS date+time and renamed the logfile accordingly.

Regards,

Stefan

meee1, 2016-05-30:
what software and version is on the autopilot?

t2000, 2016-05-30:
My Copter is running ArduCopter 3.2.1

meee1, 2016-05-30:
could you please provide one of the bad logs?

I assume you had gps lock. as the old code for downloading old logs via terminal has not changed in some time.

t2000, 2016-05-30:
Yes I did have a a GPS lock. I can prove you a log file this evening (german time).

Do you want the bin file or the plain text one?

meee1, 2016-05-30:
bin is normally easier. thanks

t2000, 2016-05-30:
I have attached one of the log files that was downloaded with the wrong name, but contains correct GPS information.

[4 1-1-1970 1-00-00 AM.bin.zip](https://github.com/ArduPilot/MissionPlanner/files/289867/4.1-1-1970.1-00-00.AM.bin.zip)

Before the MissionPlanner Update from 1.3.37 to 1.3.38 it worked fine with my Mini APM running Arducopter 3.2.1

kraftor, 2016-06-01:
I have the same problem with naming in 1.3.38 under Mono on Linux. Have not yet tried on windows. In addition I believe logs are somehow reading past the end. I see non linear GPS MS where it looks like older logs even though I clear, are, if longer than present log appended to the end. 
I will add a logfile when I get to the computer they are on.

kraftor, 2016-06-01:
I have attached the .bin file. You will see at the end of the log, there are remnants of two other previous and slightly longer flights. Also, as mentioned, none of the files are renamed (thus get overwritten each time). I noticed this start with 1.3.38 as the interface for downloading also changed (now giving a progress bar instead of byte count.)
[1 12-31-1969 4-00-00 PM.zip](https://github.com/ArduPilot/MissionPlanner/files/293926/1.12-31-1969.4-00-00.PM.zip)

kraftor, 2016-06-01:
At the risk of too much information. I have pulled the same APM log from mission planner 1.3.37 and 1.3.38 under windows and 1.3.38 under mono on ubuntu. I confirm that 1.3.38 does not rename logs or generated files under 1.3.38 linux or windows. 
Perhaps I should file the extended log issue I see under Linux as a separate bug. In the zip is two screen shots of these same logs, one from linux, one windows. You can see the linux has the previous flight appended (the log was for a 3 min flight)
[mpissue.zip](https://github.com/ArduPilot/MissionPlanner/files/294105/mpissue.zip)

meee1, 2016-06-01:
how are you downloading the logs?
using the terminal? or df over mavlink?

kraftor, 2016-06-02:
I am downloading with a usb cable via the Mission Planner datalog when I
click on "Download DataFlash log via Mavlink" button.
I notice a lot of CRC errors. It may be nothing new, but I had not noticed
it before. Tried different USB cables just in case, but no difference.

t2000, 2016-06-02:
I am also downloading the logs via mavlink, because the firmware i am using on my copter does not have a terminal anymore due to space restrictions on my Mini-APM.

@kraftor already mentioned that he uses mono under linux, but it makes no difference to the problem compared to windows. I am using mono under linux too.

This bug must have been introduced between 1.3.37 and 1.3.38 and I could imagine that it has something to do with the new download interface which now shows a percentage bar (which goes >100 ???) instead of the downloaded bytes.

meee1, 2016-06-17:
fixed here c24087a039de14c5c4773b305b78c223cad25c3d

closed 2016-06-17 by meee1