ArduPilot/MissionPlanner #1270: Show only hardware compatible version while updating firmware

misiek08, 2016-05-30:
I have a request for MP. Is it possible to make it show only compatible FW version. I connected APM 2.6 today, went to Firmware tab in MP and it shows v.3.3.3 available to download. I think MP should prevent downloading incompatible software after detecting hardware version (of course if it's possible and I think it should be).

meee1, 2016-06-01:
mp does prevent downloading incompatible software. however it is a chicken and egg scenario. as i cant display a version number without knowing the hardware first.

at this point in time i dont plan on changing the current operational state, as for current generation hardware the default versions are correct.

please upgrade to a pixhawk class autopilot.

misiek08, 2016-06-01:
I understand all of that, upgrade will be made but it's expensive stuff for
me right now. I just thought about showing some info out just disabling
buttons when MP gets info from hardware.

Can be closed.

closed 2016-06-03 by meee1