ArduPilot/MissionPlanner #1271: Bin DataFlash log review missing fields in v1.3.38, the converted .log file works

magicrub, 2016-05-31:
Bin log file missing fields. Convert same bin file to log file and when reviewing that file the fields are populated.

Try this log: https://drive.google.com/open?id=0B6RPB3yIS_0MM3piUExCN0ltelE
review the bin, check any TECS field. Then convert to log, check same TECS fields.

meee1, 2016-06-01:
looks like its a missing DF packet.
the FMT TECS message is missing from the .bin file, and is why it is dropped.

the reason the log contains it is its using a previously processed logs fmt message.

so it does highlight an issue in that the fmt is missing, and technically the .log should not have decoded the tecs message at all either.

magicrub, 2016-06-01:
Interesting

meee1, 2016-06-01:
ok looks like it is in the .bin file.
but on line 483710, not in the header as one would exepect.
FMT, 254, 64, TECS, QfffffffffffffB, TimeUS,h,dh,hdem,dhdem,spdem,sp,dsp,ith,iph,th,ph,dspdem,w,f

so this just gets weirder

so there are TECS messages, but the only appear after line 483710

magicrub, 2016-06-01:
Ok, yeah that's intentional.  It's a quadplane flight so the tecs
controller did not kick in until we went from qhover to normal forward
flight.

meee1, 2016-06-03:
fixed in current beta. now scans for all fmt's across the entire bin.

closed 2016-06-03 by meee1