ArduPilot/MissionPlanner #1275: RFD900+ - New 2.0 SiK firmware doesn't populate GUI correctly

gmorph, 2016-06-04:
I have installed the latest v2.0 SiK firmware on my RFD900+ (via Linux and the upload.py script).  I then connect it to MP (1.3.38) and navigate to "Initial Setup|SiK Radio" and click "Load Settings".  Everything seems to work correctly but the GUI doesn't populate with the correct values.  I have attached a screenshot showing the issue in MP.  I also have the MP console open and have attached a screen shot showing that it seems to retrieve the settings and information correctly but isn't populating the GUI with the information.
Let me know if I can help further.
Thanks, Grant.

[image]

[image]

meee1, 2016-06-04:
correct, please run beta MP.

this is because the S? numbers changed.

gmorph, 2016-06-04:
I have tried the beta.  The Beta version I get is 1.3.38 build 1.1.5999.29001.  I get exactly the same result.  To upgrade to the beta I went to "Help" and clicked on "Check for BETA Updates".
Happy to test anything.
Thanks, Grant.

meee1, 2016-06-04:
I just knoticed that there is no S on the start of the id's in the screenshot.

I think the firmware should be fixed.

gmorph, 2016-06-05:
Ok - I'll have a look - thanks Michael.

gmorph, 2016-06-05:
A new Beta firmware has been created that resolves this issue.  This wasn't an MP issue.  Thanks Michael!

closed 2016-06-05 by gmorph