ArduPilot/MissionPlanner #1276: Request for Security Contact

AndersAbel, 2016-06-04:
I've found a security issue in the code that I'd like to report privately. Please mail me for details.

magicrub, 2016-06-04:
Im guessing its the Mavlink1 stuff which is addressed in Mavlink2 with
packet signing.

magicrub, 2016-06-04:
To help narrow down the list of contact people, do you know if the security
issue is with the autopilot or gcs or comms link?

AndersAbel, 2016-06-04:
Trying to check how it is used and it is in a utility class that is internal, which does not appear to be used anywhere. Is there any risk that there is code outside of this repository that can access classes that are internal in this repository? If not, it's old code that is not used.

I didn't check the usage before, only noticed a known pattern that leads to vulnerabilities.

meee1, 2016-06-17:
I removed what I believe to be the issue. (tracing GitHub user to your website. etc. etc.)

AndersAbel, 2016-06-19:
@meee1 You're guess was right. The issue was in the verification code for signed xml, where [references were not checked](https://coding.abel.nu/2015/12/xml-signatures-and-references/).

closed 2016-06-19 by AndersAbel