ArduPilot/MissionPlanner #1277: Enhancement - create mission for vertical mapping

Fireflyer451, 2016-06-04:
Please add the ability to auto-generate mapping missions that allow images to be taken in the vertical plane. As in mapping a cliff or quarry's vertical face.

jagrandle, 2016-11-06:
+1 This would be a great enhancement to have. Not sure if this is the right place but it would be great to also include an ability to set a horizontal stand off distance from an object that may be sensed by a LIDAR or any other type of range sensor.

martin-git, 2016-11-23:
+1 Would be extremely helpful for geological/structural mapping of vertical structures! There is just no good way so far to do anything like it so far.

aercamti, 2016-11-24:
@jagrandle  that is a very good point. 

I have a similar requirement for a constant distance based on sensor readings - but in a horizontal plane - and outline it here http://discuss.ardupilot.org/t/need-to-create-auto-mission-which-at-each-waypoint-descends-to-height-x-agl-with-sensor/12604

Yet, if I understand correctly, Mission Planner may currently have a limitation built into the waypoint format: "mission planner only supports a single mission frame per mission" the developer Michael Oborne pointed out the above mentioned thread.

martin-git, 2017-03-11:
Hm, just read it. But as I understood it, the "single mission frame" has no real impact on a vertical mapping feature. 

> "Single mission frame", I take it, means that what Bill has outlined: Either terrain based or abs/relative - correct?

That just means during one mission everything has to be either in absolute, relative or terrain based coordinates. Correct me if I am wrong. 

Anyways, it would still be a killer feature (at least from an engineers/architecture standpoint)

aercamti, 2019-02-23:
Here is an example of how this feature has been implemented in another gcs: https://www.youtube.com/watch?v=zr7i3z60r8w 
Hope this can serve as an illustration.