ArduPilot/MissionPlanner #1279: Additional CompIDs on vehicle causes MP to report dropped packets

tpascaru, 2016-06-09:
Besides the APM, we are using an on board processor to send MAVLINK to our GC. The APC MAVLINK compID is set to 1, we would like our processor compID set to 2 (so we can pull non-APM messages out of the stream on the ground).

It seems that Mission Planner will only count a packet as "valid" if the sequence number increments AND the component ID is 1. MP reports compIDs not equal to 1 as lost packets, even if they are properly sequenced. We confirmed that changing our processor compID to 1 from 2 results in mission planner NO LONGER reporting errors. 

Is this behavior by design? It seems that the Mavlink stream quality should NOT be related to alternate component IDs in the stream.

meee1, 2016-06-09:
MP treats each component as having its on sequence stream. ie every device that has the ability to send packets maintains its own sequencing.

tpascaru, 2016-06-09:
OK -- we can adjust to that method. I was concerned that MP solely looked at the sequence counter without regard to the component ID to determine lost packets. I'm going to close this -- thank you for the clarification

closed 2016-06-09 by tpascaru