ArduPilot/MissionPlanner #1282: Tracker: add support for ALT_SOURCE, SERVO_YAW_TYPE, SERVO_PITCH_TYPE

rmackay9, 2016-06-10:
Tracker v0.7.8 (current in beta testing) has support for a few new fields.
ALT_SOURCE : allows user to control whether pitch axis is driven by the vehicle's GPS (or more accurately EKF) altitude or the barometer's.
SERVO_TYPE has been replaced by two new parameters, SERVO_YAW_TYPE and SERVO_PITCH_TYPE.
In the picture below I've also moved the pitch axis's slider as mentioned in this other to-do request.  https://github.com/ArduPilot/MissionPlanner/issues

[image]

meee1, 2016-06-17:
assume is good

closed 2016-06-17 by meee1

rmackay9, 2016-06-18:
Thanks very much!