ArduPilot/MissionPlanner #1283: expanding Wapypoint Columns

IronDome2, 2016-06-10:
Good day 

I'm sure this is not a new issue but I'm suggesting if possible to allow the Waypoint table columns to expand. Apologies if someone else has already suggested this. See image attached, the columns under red bracken can not be adjusted to widen them as need.

Thanks in advance
ID
[image]

meee1, 2016-06-10:
currently all those cells are set to autosize, based on content.
if you click inside one of the lat/lngs it will dynamically change.

IronDome2, 2016-06-10:
Hi

I did that before and I have done again but nothing happens.
Why not allow users to expand them?

Thanks for your dedication and time!

meee1, 2016-06-10:
because I cant do both. and I prefer the automatic version. how did you get into this state? load a wp file?

IronDome2, 2016-06-10:
I both ways. I created WPTs and tried to have that expanded as I wanted to check my WPTs did not work. Also I tried with loaded WPTs same thing. Actually I just installed the latest version which is what I have now and does not seems to expand. Does it work on windows 10?

By the way, how do I automatically  "DO_LAND_START"for multiple segment autonomous flights. i have mission plan with over 5 hours flight which I want to break into segments but the landing does not get added. Sorry for asking many questions in same thread!

ID

iskess, 2016-06-12:
Please see issue #941. For the past couple years I have had the same request for the opposite reason. My GCS screen is too small to fit all the columns. During flight, the Lat/Longs are not useful information. My first choice would be to allow me to rearrange the order of the columns, so I can put the lat/longs out of view off the right side of the screen. This would allow me to bring the Gradient and Altitute into view.  
The 2nd option is to allow my to adjust the column with so I can shrink the Lat/long column down to being nearly invisible. Maybe a double click on the column header toggles between autosize and hidden. 
There is definitely some room for some options here.

IronDome2, 2016-06-12:
I'm glad that I'm not alone. What I have notice the auto adjust sparingly works, most of the time it does not. So the best is to allow users to adjust as they want. The double click autofit is a good idea, just like excel, right?? In addition, by the way may this deserves a separate thread, but I find also the graphics not so user friendly, a bit clunky. Distance measuring tool is pretty awkward that you have to right click twice just to measure distance. I guess some small tool bar for these kind of quick tools would be really handy. I hope this is not too much for the developers.
Thanks once again.

iskess, 2016-06-12:
See my request #223 also from 2 years ago regarding the Distance Meauring Tool.

IronDome2, 2016-06-12:
Waaw!! Thanks iskess.

IronDome2, 2016-06-12:
I think another really useful feature would be the option to move and resize the polygon without having to edit polygon points individually.

iskess, 2016-06-13:
You really need to start a new issue for each request. I'm not clear on what you mean by the way. I can drag the polygon points and resize the polygon. You can move multiple waypoints by using the CTL button to select multiple. I can't remember if this functionality works for Polygon points as well. I'm not near a computer to try.

IronDome2, 2016-06-13:
Ok thanks for the answer. But if the polygon is a rectangle or square it would be more intuitive being able to resize or move the polygon by simple dragging from one of the polygon's corner. At the moment I have to adjust each point individually. It is perhaps hard o make too many demands, it is free software. Thanks for all.