ArduPilot/MissionPlanner #1284: Flight Plan Altitude precision (decimals)

juandiazb, 2016-06-11:
When creating a Flight Plan with waypoints, if altitude is set with decimals it is not handled correctly when saving to a file (for example 47.6 meters or feet):
- When a WP with altitude 47.6 is saved to file, the decimals are saved correctly to the file. If I open the file with a text editor, I see 47.6.
- When a file with altitude with decimals is loaded again, the decimals are not loaded, in Mission Planner, it will show 47 (if saved again to a file, it will store 47).
- The column angle (vertical angle between one waypoint and the next) ignores the decimals also. If I set altitude 47.6, it will show the same angle as if altitude was 47.
I do not know how this decimals are handled when uploaded to the Pixhawk.

meee1, 2016-06-11:
the rounding was an old limitation. which I have now removed.

juandiazb, 2016-06-11:
Great to hear! thank you very much for the quick support.

closed 2016-06-17 by meee1