ArduPilot/MissionPlanner #1287: Enhanced Camera Trigger Geotagging

cczeets, 2016-06-14:
First of all, Mission Planner rocks!! As it has since I started using it years ago!!

I'm in the process of testing out the enhanced camera trigger with RTK GPS. To check the viability of not using Ground Control. I have a highly accurate test location setup with around 20 control points set with mm accurate survey equipment.
I am using the standard 3DR GPS for Navigation and a Piksi as GPS2 for logging and processing after the flight (mainly because the Piksi RTK is too flaky for flight navigation). I'm currently flying with a Sony A7 using the Tuff Wing Hot Shoe cable. Everything works great!!
My only issue is how do I geotag the images with the precise lat, long, alt position of GPS2 and the roll, pitch, yaw?
The CAM messages that are created correspond to GPS1 and the sampling rate of my GPS2 isn't fast enough to log at the speed I'm flying (~7m/s), unless I interpolate manually. Can I get the CAM message to use GPS2 instead of GPS1? Or is there a better way to do what I'm trying?  Also, can the geotag utility in Mission Planner interpolate based on the before/after GPS2 positions and use the GPS2 speed to get a more accurate geotag position?
This may not seem necessary, but when using such an accurate GPS it is definitely needed.
I have a Reach GPS on the way as well so I can test the two against each other. I know they have a Hot Shoe setup that will tag the RINEX file when the camera fires as well. I'm planning to use that for post-processing.  I will post my results once everything is complete and processed.

Thanks!
Steve

cczeets, 2016-06-14:
Sorry, one more thing to add.  It would also be nice if there would be a place to put in offsets from the GPS2 to the Camera Focal Plane (x, y, z as related to the autopilot orientation?).  These could be taken into account when interpolating the position to get a more exact location of the camera when triggered.  Especially if we want to get down to cm or mm precision...

Thanks again!
Steve

meee1, 2016-06-17:
the bigger issue here is lever arm correction. as the offsets from the gps change with roll, pitch and yaw.
currently there is no method to do what you are asking. however the piksi does support event inputs. ie hotshoe

DroneDOOD, 2018-05-17:
I see this is almost two years old.  But I am wondering if anything has moved forward with lever arms.  I think GPS2 is now an option.