ArduPilot/MissionPlanner #1288: Live calibration seems to be broken for second compass

staroselskii, 2016-06-15:
FS: ArduCopter 3.4-rc1
Hardware: Raspberry Pi 2 + Navio 2

Our users have spotted a strange issue with calibration offsets. 

Take a look at these pictures:
1) Live Calibration

Offsets after the calibration:
[image]

Tuning tab:
[image]

2) Onboard Calibration
Offset after calibration:
[image]

Tuning tab:
[image]

It's pretty evident that something's wrong with ```my``` and ```my2``` being so different. In fact it seems as if OFF2 was inverted the whole thing would've been gone. To confirm I adjusted the parameter manually and it did the trick - all axis got intact and aligned. So I suppose 
I was also a little confused with the fact that offsets don't get updated once onboard calibration is complete. It seems like the Y axis is to be inverted.

A quick glance at MagCalib.cs hasn't resulted in a solution, so I'd be extremely glad if someone could have a look.

meee1, 2016-06-17:
are you sure you have the correct orientation set for compass 2?

staroselskii, 2016-06-17:
@meee1 

Yes. Pretty sure. Some of our users have confirmed that onboard calibration works and the live doesn't (with orientation not changed between the tests). Although I do share your scepticism, I'm no static analyzer. Is there a chance the could be a typo in this piece of code?

ivereninov, 2016-07-15:
Is there anything we can do to help debug? The issue has been confirmed by several users.

staroselskii, 2016-11-21:
Sorry for the hustle. It was my bad all the way. Will be fixed with https://github.com/ArduPilot/ardupilot/pull/5249

closed 2016-11-21 by staroselskii