ArduPilot/MissionPlanner #1289: APM:Plane 3.6.0 dataflash log format change has broken georefimage

nicksargeant, 2016-06-17:
The Georef tool no longer works with dataflash logs generated from autopilots running APM:Plane 3.6.0.
[image]

 3.5 GPS string: FMT, 130, 46, GPS, QBIHBcLLefffB, TimeUS,Status,GMS,GWk,NSats,HDop,Lat,Lng,Alt,Spd,GCrs,VZ,U
3.6 GPS string: FMT, 130, 50, GPS, QBIHBcLLeeEefB, TimeUS,Status,GMS,GWk,NSats,HDop,Lat,Lng,**RAlt**,Alt,Spd,GCrs,VZ,U

nicksargeant, 2016-06-17:
There also appears to be an issue with Copter 3.4-rc1

magicrub, 2016-06-17:
That data is found in other fields. Instead of ardupilot logging redundant
data could this tool simply parse the other fields?

nicksargeant, 2016-06-17:
Agreed. The inconsistency and duplication within the log structure is frustrating. For example, why is rangefinder distance stored as RFND.Dist1 on Plane and CTUN.SAlt on Copter?

meee1, 2016-06-17:
could I get a dflog for this one? MP does handle RAlt/RelAlt fields so wondering if its hitting some other issue.

nicksargeant, 2016-06-20:
Sure Michael, email sent! Thank you as always :)

voidmmn, 2016-06-20:
I had the same problem, if you want I have a lot of dflogs.

nicksargeant, 2016-06-23:
Hi Michael,
Any is there enough information in the logs to get a sense of the issue? For now, do you recommend sticking with APM:Plane 3.5?

meee1, 2016-06-23:
I never got an email with the log.

could you try MP beta? and tell me what it says in the status box? it should list the number of point extracted from the log, as well as the image count.

nicksargeant, 2016-06-23:
Perhaps i have the wrong email. I wont pubicly list your email address but can you please send me an email to nick.sargeant@gus-uav.com/gmail.com andI will reply?
I have attached the outputs for copter3.4 and plane 3.6.
Cheers,
Nick
[plane36.txt](https://github.com/ArduPilot/MissionPlanner/files/329362/plane36.txt)
[copter34.txt](https://github.com/ArduPilot/MissionPlanner/files/329363/copter34.txt)

nicksargeant, 2016-06-25:
It is working with the latest beta, thanks Michael!