ArduPilot/MissionPlanner #1291: Geotagged images not geotagged

Jman841, 2016-06-17:
When I use the geotag image function in Mission planner, it successfully makes a .txt file with the image name and geo tags, however, when it processes the images into the new geotagged folder, none of the geotagged metadata is attached to the image, at least as far as I can see and it does not load in Pix4D.

meee1, 2016-06-23:
I think I need more info, nothing has changed from MP side to do with this, so I don't have anything to go by.

Jman841, 2016-06-23:
Sorry,

here' is 3 of the original photos I tried to geo tag, the files produced showing that it found the geotag information in the .txt and .kml files, and then 3 files that were supposedly geotagged in the metadata by mission planner, however, no metadata was added...

https://www.dropbox.com/sh/crhovsi4hkaljd2/AABC63xy2W936fcjmJo8CSmda?dl=0

Jman841, 2016-07-12:
I've done a bit more digging into this, the .txt file and .kml show the photos with the geotags properly assigned, however, the .csv file is blank and there is no data added to the processed photos. Is it possible that when the app in mission planner applies the geotags it uses the .csv file and for some reason something is broken where the .csv file is not being properly created?

Jman841, 2016-07-12:
I reprocessed the images as .jpeg and reran the tool and it tagged everything correctly. I guess the issue is .tif/.tiff files are not supported by the geotagging tool in mission planner. This is important as a compressed jpeg file can loose quite a bit of the calibrated data from some of these multispectral cameras. Could we add support for tagging of .tiff files and other file types?

Thank you!