ArduPilot/MissionPlanner #1293: Loiter radius not shown in planner

Fossdal, 2016-06-21:
Loiter radius is not shown i planner only wp radius

meee1, 2016-06-21:
this is by design. please use one of the other screens to set loiter radius.

meee1, 2016-06-23:
just to clarify.
plane shows loiter radius.
copter does not.

Fossdal, 2016-06-23:
Just to clarify.
i am using Plane
In "Flight Plan" if i change loiter radius nothing changes on the map, if i change the wp radius. the grafical circle is changed on the map both for loiter waypoints and normal waypoints

meee1, 2016-06-23:
the map is not meant to change if you change loiter radius. do you have loiter commands in your mission?

magicrub, 2016-06-23:
just to clarify, WP_RADIUS has nothing to do with this.
http://ardupilot.org/plane/docs/parameters.html#wp-radius-waypoint-radius

WP_RADIUS is poorly named and really needs a rework. See https://github.com/ArduPilot/ardupilot/issues/3483

Contray to popular belief, setting WP_RADIUS to 200 does not mean you "reached" the way point at that point, it means the aircraft is allowed to decide when it reached it and will allow it to do a smooth controlled turn somewhere within that radius.

Naterater, 2019-01-04:
I think the logic and viewing of waypoint radius and lotier radius are adequately displayed currently. Even the exit tangent displays correctly. Please close.

magicrub, 2019-01-04:
Hold on a second, I think this is still useful. @DavidIngraham got any ideas about this one?

Naterater, 2019-09-30:
Close?