ArduPilot/MissionPlanner #1294: Freeze on huge/invalid float in MAV_MSGID_VFR_HUD

BoskiDialer, 2016-06-23:
Mission Planner 1.3.38 build 1.1.5983.12141

When i send MAV_MSGID_VFR_HUD with airspeed=2.147E+9, mission planner works properly, but when i send airspeed=2.148E+9, mission planner starts using 100% cpu - i am able to freeze mission planner by sending invalid message from object.

Same with groundspeed: 2.147E+9 is ok, 2.148E+9 freezes.

This looks like casting to int32, because 2.147e9 < int32_max, but 2.148e9 > int32_max. I can also freeze it by sending +0.Infinity.

Negative values like -1e30 and -0.Infinity are fine.

alt and climb in MAV_MSGID_VFR_HUD are not affected.

meee1, 2016-06-23:
can you provide the tlog? i asume it made it to that

BoskiDialer, 2016-06-23:
tlog, rlog and MissionPlanner.log

[logs.zip](https://github.com/ArduPilot/MissionPlanner/files/329406/logs.zip)

meee1, 2016-06-23:
new beta uploaded that should resolve this

closed 2016-06-23 by meee1

BoskiDialer, 2016-06-23:
that was partial fix

Mission Planner 1.3.38.1 build 1.1.6018.28293:
1e9 ok
2.1e9 ok (ticks on the left[hud] start dissapearing)
2.2e9 ok (same as above)
4e10 not freeze, value on the left[hud] is -2147483648, down left corner says AS 40000000000,0
1e12 (0x5368D4A5) mission planner starts to lag
1e13 (0x551184E7) mission planner freezes, but when value changes, it manages to unfreeze
1e14 (0x56B5E621) freeze
1e20 (0x60AD78EC) ok, but AS not updating
1e30 (0x7149F2CA) ok, but AS not updating
0.Infinity (0x7F800000) freeze