ArduPilot/MissionPlanner #1295: extract params from log file

magicrub, 2016-06-23:
Maybe another task for the create KML+gpx button? Maybe its time to rename that button label? :)

feature request: extract params from .bin/.log file. method options:
- one file with all params
- One file for the boot-up, another file for all further changes (similar to mission extraction)

cala2, 2016-10-16:
+1 create a param file with a log one, usefull for upload to vehicle.

magicrub, 2019-05-29:
done a long time ago

closed 2019-05-29 by magicrub

ddomit, 2021-08-04:
> done a long time ago

How can i do this? Can you please explain? I dont know where i can extract the .param file from the .bin file

magicrub, 2021-08-04:
@ddomit This is hiding in the "create KML+gpx" button and needs a better label, can you add that @meee1 ? That button also extracts missions and mission changes.
[image]

ddomit, 2021-08-05:
Thank you!! @magicrub