ArduPilot/MissionPlanner #1297: Add firmware upload for Bebop2

rmackay9, 2016-06-24:
It would be good to add support for uploading firmware to the Bebop2.

Instructions are in the "Uploading the firmware" section of this wiki page: http://ardupilot.org/dev/docs/building-for-bebop-2.html

There are some tricky bits:
- adb needs to be installed on the user's machine
- we need to modify some scripts on the bebop2 and we want to be able to backout those changes as well so that the default bebop2 firmware can run.

@OXINARF had some ideas about how this could be done.  I've got a Bebop2 and so does Leonard so we'd be happy to test.  If we're going to do this it may be possible to get Parrot (maybe we could even convince @jberaud to send a Bebop2 to the developer who takes this on?).

jberaud, 2016-06-24:
Yep OK to send a Bebop for this. It's going to be needed for Disco also at some point.
adb is needed. There are other solutions regarding the startup script though they would all involve modifying the startup scripts via adb.

OXINARF, 2016-06-24:
I could take the job but I can only commit to work on it at the end of July.

meee1, 2016-06-28:
couple of questions.
1. is the bebop pingable as soon as the wifi is connected?
2. is port 9050 open if adb is not running?

jberaud, 2016-06-28:
1. yes
2. What do you mean by "open" ?

jberaud, 2016-06-28:
provided a program is listening on the bebop's side yes but there is nothing listening on that port apart from adb.

meee1, 2016-06-28:
so I assume the power 4 times trick just starts adb? its not just a firewall mod or something similar.

ie just trying to establish when the bebop is connected, and will accept a adb connection, without actually starting a adb session and waiting for the timeout.

jberaud, 2016-06-28:
>so I assume the power 4 times trick just starts adb? its not just a firewall mod or something similar.

True
>ie just trying to establish when the bebop is connected, and will accept a adb connection, without actually starting a adb session and waiting for the timeout.

what can be done is connecting the bebop via usb. When pressing the button 4 times, you could just wait for an rndis interface to be present and the DHCP to have succeded, then adb is available.

rmackay9, 2016-06-28:
oh wow.  I didn't even notice that there was a small usb connector on the back of the Bebop2 right below the red button.  If connecting that would make it easier for @meee1 to implement the firmware upload, from a user point of view that would be a-ok (i.e. no need for it to be over wifi really).

meee1, 2016-06-28:
@rmackay9 
can you test this via usb (i asume you have the adb usb driver already installed)

adb -d shell mount -o remount,rw /
adb -d push arducopter /usr/bin/
adb -d shell cp /etc/init.d/rcS_mode_default /etc/init.d/rcS_mode_default_backup
adb -d push rcS_mode_default /etc/init.d/
adb -d shell sync
adb -d shell reboot

meee1, 2016-06-28:
"adb devices" should list one device.

jberaud, 2016-06-28:
>adb -d shell mount -o remount,rw /
"adb remount" works too

jberaud, 2016-06-28:
@rmackay9  
>I didn't even notice that there was a small usb connector on the back of the Bebop2

makes it very convenient to control it with a companion computer

rmackay9, 2016-06-29:
I don't really know what I'm doing I think.  I tried connecting the Bebop2 to my Windows 10 PC using a USB cable.  It appeared as an "Other Device - RNDIS".
I have adb on my machine so I then tried to run "adb usb" to make it connect over USB.  it produced output like this:
_C:\Users\rmackay9>adb usb
* daemon not running. starting it now on port 5037 *
* daemon started successfully *
error: device not found_

I then tried installing 4 different drivers.  3 of them I downloaded from google: https://developer.android.com/studio/run/win-usb.html#top  (see "Click here to download the latest Google USB Driver ZIP file.").
This gave me three to choose from, "Android ADB Interface", "Android Bootloader Interface" and "Android Composite ADB Interface".  I tried them all (and they could successfully be installed) but the result of the "adb usb" command was always the same as above, "error: device not found".
I uninstalled the above drivers and then tried installing the windows network interface as outlined on this web page: http://developer.toradex.com/knowledge-base/how-to-install-microsoft-rndis-driver-for-windows-7... but again the result of the "adb usb" command was the same.

So really I have no idea how to create an adb connection using the usb cable.  Any ideas?

jberaud, 2016-06-29:
@rmackay9 What ip address do you have on your rndis interface ?
If it is 192.168.43.XXX you just have to type "adb connect 192.168.43.1:9050"
It uses an rndis connection and not adb over usb.

rmackay9, 2016-06-29:
I don't think my rndis interface is working.  I'll google around for drivers for windows 10.  That link in the previous comment (related to windows 7) didn't work.
[image]

jberaud, 2016-06-29:
@rmackay9 what does "update driver" does ?

rmackay9, 2016-06-29:
I'm afraid it doesn't find any compatible drivers.

jberaud, 2016-06-29:
oops... I am sure it can be done though. I'll try to look around too.

OXINARF, 2016-07-04:
Since using USB you still have to press the button four times, what is the advantage of using it over Wifi? And apparently users could get driver issues?

My plans have changed and I could actually work on this now, but I'm not sure how much @meee1 has worked on it (and he also knows much better the Mission Planner code structure and C# than I do).

rmackay9, 2016-07-05:
@OXINARF, I think there's probably no real advantage to using a USB cable except that the user must connect their wifi to their bebop2.  This means, on a laptop with only a wifi connection, it will be awkward to do the firmware download and the connection to the Bebop2 at the same time.
I'd take whatever I can get at the moment.

OXINARF, 2016-07-05:
@rmackay9 Sure, there might be some quirks to iron out regarding user interface when using Wifi but I think it can be doable.

Like I said before I can take this, but if @meee1 is committed to do it there's no point in me doing it. I would need to take the offer from @jberaud for a Bebop2 and obviously I'm not going to request that if I'm not the one doing the job.

I'll leave you three, @rmackay9, @jberaud and @meee1, to decide what to do.

meee1, 2016-07-05:
@OXINARF fell free to take up the offer and work on this if you want. I hadn't gotten that far as I was still evaluating what needed to happen.

jberaud, 2016-07-06:
@OXINARF I am ok to send a bebop2 to you or the one doing the job. We should just discuss how this is to be done, since some things might change on the future firmwares and we have to make sure your procedure wouldn't be broken by a soon-to-come update.

OXINARF, 2016-07-06:
@jberaud That's fine. Would you like to discuss that here or another place (Gitter, Skype)?
Also I would like to know what you think would be a good timeline to have this done.