ArduPilot/MissionPlanner #1298: HUD not updating when waiting for ACK to CMD_LONG/PREFLIGHT_STORAGE

BoskiDialer, 2016-06-24:
Mission Planner 1.3.38 build 1.1.5983.12141
Mission Planner 1.3.38.1 build 1.1.6019.12089

My controller is under development, so it currently ignores MAV_MSGID_COMMAND_LONG.

If i change some parameters ("config/tuning" -> "full parameter list" -> "write params") in T+0s, controller receives changes and responds to them, Mission Planner works fine.
In T+10s, Mission planner sends MAV_MSGID_COMMAND_LONG with command=MAV_CMD_PREFLIGHT_STORAGE, but due to state of controller, it is not responding (same may happen when packet is lost). Mission planner resends this command at T+12s, T+14s and T+16s. HUD is not updating from T+10s to T+18s.

HUD is also not updating when:
- MissionPlanner is trying to update home location
- In "actions" i send "MISSION_START" -> for ~8s
- In "actions" i send "LOITER_UNLIM" -> for ~8s

meee1, 2016-06-24:
cmd_long in mission planner have been internationally blocking commands. as there can be large state change based on the command.
i will be fixing the preflight storage issue, as that was only there for older compatibility.
as for the other, i will need to look harder

meee1, 2016-06-28:
these should all be fixed now.
feel free to reopen if they are not.

closed 2016-06-28 by meee1